tedsalmon/BlueBus #2: [CD53] BlueBus displaying text when it shouldn't

dutchhome, 2019-02-16:
BlueBus is writing to the screen even when the "CD Changer" is not selected.

Some scenarios include:
* Turning off the radio should display the clock.  Instead, "Paused" is written.
* When the radio is on (playing actual radio stations), the BT device is paused.  If you attempt to play the music on the device, the radio display changes to show the song and then shows "Paused".  The radio station is no longer displayed.
* When in CDC mode and BlueBus is playing a song, it is impossible to switch to clock mode (clock button) and impossible to modify tone settings, etc. because BlueBus continues to write to the display, overwriting these modes.

tedsalmon, 2019-02-17:
@dutchhome,

Thanks for the report! The first two issues should be pretty easy fixes and I think I know what's wrong already. As far as the inaccessibility of the clock and settings, I will have to rig up my fake IKE to broadcast the time, cause I've never had that on my CD53 :).

Thanks!
-Ted

tedsalmon, 2019-02-18:
@dutchhome,

This has been resolved in firmware version 1.0.1

Thanks!
-Ted

tedsalmon, 2019-02-18:
@dutchhome,

I did forget to mention that I implemented the "2" CD selection in order to disable (and re-enable) metadata scrolling. This is the only way to prevent BlueBus from writing over the time.

Thanks!
-Ted

closed 2019-02-18 by tedsalmon