tedsalmon/BlueBus #3: [CD53] Manual mode does not work

dutchhome, 2019-02-16:
When switching to manual mode (pressing the "m" button between the prev / next buttons), the prev / next buttons should switch to rewind / fast forward.  Instead of passing these commands to the bluetooth device, the buttons are non-functional.  Manual mode must be turned off in order for prev / next to work again.

tedsalmon, 2019-02-16:
@dutchhome,

Thanks for the report! The BC127 doesn't have "seek" functionality, unfortunately. I will see what the CD53 broadcasts over the IBus when set to manual mode (I had no idea what the M button was for, as I've only ever had E46s with the nav) and continue using those for Prev / Next.

Thanks!
-Ted

dutchhome, 2019-02-17:
According to this site, it seems to support it:

https://learn.sparkfun.com/tutorials/understanding-the-bc127-bluetooth-module/all

> A2DP is a two-channel audio streaming protocol, which transfers data from a source to a sink. The source can be a cell phone, a media player, or the audio input on another BC127 (or similar) module. AVRCP allows the BC127 to send commands to the remote device which will cause the remote device to play, pause, stop, advance or reverse playback; some remote devices will transmit track meta data back to the BC127 via the AVRCP protocol.
> 
> The BC127 also supports the Hands-Free Protocol (HFP); we’ll talk about that later, but for now be aware that incoming calls over the HFP connection will override data on the A2DP connection.
> 
> Here are the commands you’ll need to know to get A2DP and AVRCP working properly:
> 
> MUSIC instruction sends a command over the AVRCP link to the A2DP source device. Valid options for instruction are PLAY, PAUSE, STOP, FORWARD and BACKWARD.

dutchhome, 2019-02-17:
Sorry, I see what you mean now. Clearly I misread that section. By forward and backward it does seem that they mean next track and previous track. Yes, next track / previous track would still be preferable to non functioning buttons.

tedsalmon, 2019-02-17:
@dutchhome,

_Actually_, it looks like you're right! As of Melody 7.x, there's `FF_PRESS`, `FF_RELEASE` and `REW_PRESS`, `REW_RELEASE` commands.

[image]
I'll have to see how the radio signals these actions to the CDC to make sure the implementation will work, but it looks solid for now.

Thanks!
-Ted

tedsalmon, 2019-02-18:
@dutchhome,

This has been implemented in firmware v1.0.1 but I have noticed that at least on my Pixel 3, the seek actions have no effect on playback. As more people continue to test, I will see if this is the case for most people and simply ignore manual mode and re-implement this as Next / Previous, rather than seek.

Thanks!
-Ted

closed 2019-02-18 by tedsalmon