tedsalmon/BlueBus #16: Feature Request : Door Unlock/Lock Comfort Entry

mykekfz, 2019-10-07:
Hello Ted,

How bad will power consumption be if we were to search for paired devices every 2 secs?

I'm thinking of a comfort entry feature where if the module detects a paired device, it send the message to unlock the door the opposite happens when paired devices are not within range.

let me know what you think.

Mike

harryberlin, 2019-10-07:
Did you have tried to unlock with ibus message the car, after secure lock?

tedsalmon, 2019-10-09:
Hey Mike (@mykekfz ),

Power consumption of the unit at idle is about 50mA. It could theoretically be used to lock / unlock the car, but people like myself, who have an attached garage might run into issues as the car would constantly lock/unlock as the phone comes into proximity with the vehicle.

Futhermore, the module is designed to power off with the rest of the car and wake up after the reception of the next IBus message. This is to prevent the module from flattening your battery over time.

@harryberlin ,

I'm sorry, but I haven't actually done any testing with the Unlock/Lock IBus message(s), so I can't answer your question :(

Thanks!
-Ted

mykekfz, 2019-10-10:
Hello Ted,

Will using the command "RSSI" be helpful with scaling the proximity of the connected device?
as for battery draw, i don't think it will be an issue since the draw is so minuscule doubt it will be an issue. You can always notify the user about the consequences of this feature if they were to activate it.

@harryberlin i dont see why it will not work since the command to unlock (key press from remote) its also sent through the but i could be wrong.

@Ted is there a way/command i can use to send RAW bus message for testing?

Regards,
Mike

tedsalmon, 2019-10-11:
@mykekfz (Mike),

That might work, but I feel like it may lead to some wonky results as different cars will have the BlueBus installed in different locations which will lead to different signal strengths at different distances. Still, a very good idea!

Let me digest it a bit and see what I can come up with.

I will send you a firmware that will allow you to send messages directly to the IBus

Thanks!
-Ted

ShonkyCH, 2019-10-11:
50mA is quite high for a car although may not seem like it's that much. It's certainly not "miniscule".

BMW maximum spec is 30mA. Most are in to 30-50mA range. So an extra 50mA (assuming the worst case 30mA) is 2-3x that in total.

tedsalmon, 2019-10-11:
@ShonkyCH,

You're right and I forgot to touch on that aspect. I did the math on this for an E46 a while back and the draw would flatten the battery in about a month, which is why I decided to implement "sleep" mode where the BlueBus goes to sleep 60 seconds after the last IBus message and wakes up on the next message.

Daily driven cars would have no problem with this type of drain, but any cars that sit for more than a few days may experience issues.

Thanks!
-Ted

harryberlin, 2019-10-11:
in my case (e39), the gm don't processes diagnistic actions, if the car is secure locked.

Tomcatex, 2019-12-12:
Hello Ted,

Just had an idea and I stumbled upon this thread.
About comfort locking, can you add a feature to unlock the car, if it has been previously locked, when the key is pulled out? 
My E39 locks automatically when speed is about 20km/h. But it does not unlock when stopped.
Can you add something so that car unlocks (VW-group has this) when key is pulled out? 

Thanks

tedsalmon, 2019-12-17:
@Tomcatex,

Sorry for the delayed response. This should be pretty easy to implement, honestly.

Since the car has just been turned off, the GM should still listen for diagnostic commands.

Thanks!
-Ted

mykekfz, 2019-12-23:
ca

> 
> 
> Hello Ted,
> 
> Just had an idea and I stumbled upon this thread.
> About comfort locking, can you add a feature to unlock the car, if it has been previously locked, when the key is pulled out?
> My E39 locks automatically when speed is about 20km/h. But it does not unlock when stopped.
> Can you think of something, so that car unlocks (VW-group has this) when key is pulled out?
> 
> Cheers

Cant this be done via coding with NCSexpert?

tedsalmon, 2019-12-23:
@mykekfz,

Yes, on some vehicles the GM will allow you to code that. I'm not ready to code the GM via the BlueBus just yet, so having an option for it makes sense still :)

Thanks!
-Ted

mykekfz, 2019-12-23:
I was just thinking of MCU available space. 

If a feature is available via simple coding, I don't see the point of it adding to the bluebus. 

On the flip side not everyone knows how to use ncsexpert.

tedsalmon, 2019-12-23:
@mykekfz,

I'm not too worried about MCU space just yet -- I have 94% ROM remaining as it is :smile: 

It would be nice to allow the BlueBus to simply code the option in the GM, rather than using Diagnostic commands.

harryberlin, 2019-12-23:
it is pobble to lock the car by speed from ibus. so you don't coding.
but one point there are different messages between e46 and e39, old and new models.

tedsalmon, 2019-12-23:
> it is possible to lock the car by speed from ibus. so you don't need coding.
> but one point, there are different messages between e46 and e39, old and new models.

I noticed this as well -- looks like the E46 takes `0x97 0x01` to unlock, while the E39 takes `0x00 0x0B` to unlock. Similar differences exist in terms of the LCM bulb diagnostic messages.

Tomcatex, 2020-01-15:
> I was just thinking of MCU available space.
> 
> If a feature is available via simple coding, I don't see the point of it adding to the bluebus.
> 
> On the flip side not everyone knows how to use ncsexpert.

I have NCS. Is there a specific line for this? Can yo post any more info? 
I tried to find it, no luck. 

E39 '02 Euro 525d, LCM III.

harryberlin, 2020-01-15:
on e39 you have to log the kbus for groundmodul

tedsalmon, 2020-02-12:
> @mykekfz
> 
> > I was just thinking of MCU available space.
> > If a feature is available via simple coding, I don't see the point of it adding to the bluebus.
> > On the flip side not everyone knows how to use ncsexpert.
> 
> I have NCS. Is there a specific line for this? Can yo post any more info?
> I tried to find it, no luck.
> 
> E39 '02 Euro 525d, LCM III.
> 
> EDIT: Check this https://forum.bmw5.co.uk/topic/66086-ncs-coding-unlock-on-keyout/

[image]

For what it's worth, the above diagnostics messages are all that's available. In my E46, it appears that the EWS "key out"/"immobilized" message is what prevents the GM5 from processing any further messages. I have yet to test out artificially notifying the car that the EWS is no longer immobilized and then trying to run diagnostics.

tedsalmon, 2020-05-12:
FWIW, The BlueBus now supports comfort locks and unlock, but they are not done as OP requested due to limitations with diagnostic messages. I also can't perform diagnostics on the GM after the EWS has been armed (key removed), so there's that.

Thanks!
-Ted

closed 2020-05-12 by tedsalmon