tedsalmon/BlueBus #19: Triggering phone to "on" state to power microphone

zarboz, 2019-12-30:
https://github.com/zarboz/BlueBus/blob/5537cf12097ae68a5b5ffa108ce1afc157e5c512/firmware/application/lib/ibus.c#L185

This first line looks like a "when MFL phone is pressed THEN answer phone" 

and 

https://github.com/zarboz/BlueBus/blob/5537cf12097ae68a5b5ffa108ce1afc157e5c512/firmware/application/lib/ibus.c#L1190

appears to be only an IKE print for the telephone caller maybe? it prints to IKE TEL data.... anyway I am curious if you feel like clarifying how you triggered the phone box to "on" to provide power to the microphone. Also curious how you hijacked the signal coming from the phone box. 

Essentially my tentative plan is to feed my e38s microphone into the Node.js project I am using so that I can answer phone calls etc

tedsalmon, 2020-01-02:
@zarboz,

First off, you're correct about the first line of code -- that's simply to answer the call when the phone button is pressed on the MFL.

>appears to be only an IKE print for the telephone caller maybe? it prints to IKE TEL data
This is actually how I provide a UI for E46 "Business CD" users. The Radio listens for this message from the TEL and displays the text on its 11 character display.

You can't actually tell the microphone that it needs to be powered, you need actual circuitry to feed it 5VDC in order to power the pre-amp inside of it.

I use a 2k ohm resistor connected to 5v DC and the Mic+ line, and a bi-polar (DC Blocking) 47uF capacitor between the BT module and the Mic+ line in order to "power on" the microphone.

If you view [the 4th sheet of the schematic](https://cadlab.io/project/1479/master/circuit/aGFyZHdhcmUvQmx1ZUJ1cy5zY2g%3D) you'll see how I've done this.

Let me know if you have questions.

Thanks!
-Ted

zarboz, 2020-01-02:
did you just use positive/negative feeds for the audio pipeback into the pi?

zarboz, 2020-01-02:
Awesome thanks Ted,

It is just a physical hardware mod then from what I can tell off your board prints. I immediately went to chase down iBus commands thinking it was something you "turned on" via a command pass

closed 2020-01-02 by zarboz

tedsalmon, 2020-01-02:
The negative feed for the microphone should be connected directly to ground. You'll want to feed your positive mic lead (after you have decoupled DC through the capacitor) into whatever Mic input you have on the Raspberry Pi.

Hope this helps!
-Ted