tedsalmon/BlueBus #20: Feature Request: Output from the board

Olli-M, 2020-04-03:
Hi, I have a few things I would like to do that should be controlled via a menu and as such I was wondering if it would be possible to get rs232 out of the board or some form of GPIO so that everything could be in one module. Another hope of mine would be to be able to have features like mod light except actually in the menu and visible for navigation users. I would be happy to work on coding such features in, I just want it all to be clean and on a single module.

Another project, which is the reason I REALLY want this is I am trying to add dynamic backup lines to my e39, and need a way to constantly poll the steering wheel angle and due to the bus not having much bandwidth, it would be useful to be able to control everything from a single module.

tedsalmon, 2020-04-07:
@Olli-M,

So, The BlueBus already has a serial terminal available via the USB connector on board. Does that satisfy your needs, or do you want direct access to the UART lines?

I'm already planning on integrating the modLight2 features into the BlueBus, similar to the Intravee.

Also, The I-Bus has no access to the steering angle sensor, so unless you can get that data back to the BlueBus somehow, it's a no-go.

Thanks!
-Ted

Olli-M, 2020-04-07:
@tedsalmon 

Ok, good, that should be fine, came up with another solution, will have to get a steering wheel angle sensor and find out the protocol and use that. I hope it’s in constant communication, seems like it’s a direct link between it and the DSC controller. If not then I will need to figure out a way to communicate without overloading the DSC. Admittedly I’m a bit of a novice at all of this, but nothing that over engineering, over speccing, and over working myself can’t work through. 

In other words, just stupid enough to be so stubborn that I’ll do the impossible and make anything work. Also might be part of the reason why I got an e39 530D as my first car...

tedsalmon, 2020-04-07:
@Olli-M,

Hmm. There isn't enough space to add another header for UART. However, the PIC's UART lines are pulled up to 3.3v through some 0805 100k ohm resistors. It would be trivial to solder a couple wires to those pull up resistors, and it'd be quite easy to do (R14 and R15).

Are you sure the angle sensor isn't just a trimpot? I'm talking out of my ass a bit as I have done no research, but I don't see why BMW would've over-complicated the sensor.

Still, what are you hoping to gain by gathering this info?

Also, do you have a BlueBus? I do plan on adding caller ID when calls come in. I have to work on storing your phonebook on the BlueBus first, so I can look up the phone number sent when a call comes in.

Thanks!
-Ted

Olli-M, 2020-04-08:
Sounds good, I’ll definitely do that then.

You may not see why they would have complicated it, but they sure as hell did, it is Bmw after all. It has its own microcontroller and 6 wires connecting it to the DSC (Dynamic Stability Control) and in some cars, even the HID.

I am trying to add dynamic steering lines so that my backup camera would have lines on it showing where its turning given the way the wheels are pointing. `<img src="http://www.carnaviplayer.com/images/backupcamera/CVS-2131A.jpg" alt=“VW Backup">[image]`

I will soon, was a day late for the first batch of order, kicked myself from then till the third of April when I excitedly ordered it. Do you think the 1MB of EEPROM will be enough to get the phone book access profile to download contacts? I would have assumed that Siri could fill the role of contacts and that you could somehow get caller id or get contacts in a non statically stored way. Well, however you decide to do it, best of luck! I hope to help as soon as I get a blebs in my hands I can test and dev on, and will be happy to help and share my results.

Best regards,

Olli Massala

tedsalmon, 2020-04-09:
> You may not see why they would have complicated it, but they sure as hell did, it is Bmw after all. It has its own microcontroller and 6 wires connecting it to the DSC (Dynamic Stability Control) and in some cars, even the HID.

I guess they needed accuracy and probably yaw angle, too. But of course they complicated it... ugh.
 
> I am trying to add dynamic steering lines so that my backup camera would have lines on it showing where its turning given the way the wheels are pointing.

Got it! That's pretty neat :)

> Do you think the 1MB of EEPROM will be enough to get the phone book access profile to download contacts? I would have assumed that Siri could fill the role of contacts and that you could somehow get [...]

I plan to store the vCard name and phone number. I'll be storing the phone number as a 64-bit integer and I'll allocate 40 bytes for the contact name (so 40 characters). 48 bytes per contact will equate to about 20,000 contacts that can be stored on the EEPROM. I'll probably have a 512 contact cap per paired device. Sadly, HFP is pretty useless in terms of caller ID. All I can get is a phone number for the incoming caller :(

> Another thing that would be awesome that isn’t on the mod light would be to have various temperatures, battery voltage, current speed and such visible in both HID and the computer, that was a feature on a module called the ekombi.

This is coming :)

Thanks!
-Ted

Olli-M, 2020-04-10:
It may be pretty neat, but the more and more I am researching it, the more and more it looks like a crazy amount of work, seems like the “easiest” way to do it will be a digital camera setup, and at that point, might as well create a 5 digital camera setup, with dvr integrated into it. At that point just a matter of software to make it panoramic with a birds eye view too.

Interesting, I wasn’t considering that you could take out the rest of the details, makes sense. I would have wanted to keep address just so that there could be a command via ibus for the address (something done by the SES module, so if you can go through the contacts it would be cool to integrate that.

I’m really glad to hear that, that’s a huge hope for me.

Thanks,

Olli

tedsalmon, 2020-05-12:
Closing issue -- thanks for your feedback :)

closed 2020-05-12 by tedsalmon