tedsalmon/BlueBus #22: Odometer light on with car off after 1.1.13 firmware flash

GyroF-16, 2020-07-03:
I updated the firmware in my BlueBus to 1.1.13 yesterday, and today I noticed that the orange display at the bottom of the instrument panel (between speedometer and tach) was on when the car was parked in the garage. The odometer and trip odometer come on for 1:16 (one minute, 16 sec) then go out for 12 sec, then the cycle repeats - endlessly. I’m thinking it will certainly effect the battery over time.
Anyone else seeing this?
Could this have something to do with the BlueBus pulling coolant temp from the IBUS, even when the car is off?

tedsalmon, 2020-07-03:
I can confirm this is a problem. It appears my unit goes to sleep after 90-ish seconds, then wakes up less than 15 seconds later. I can note that the Nav computer comes back on first, so I suspect it's the state I'm leaving one of my TH3122.4 (I-Bus transceiver) pins in that is causing the issue.

tedsalmon, 2020-07-08:
This was resolved in #32311f744b0f799fb1e17a86b7a56fbf604d9891 -- Thanks for the report, @GyroF-16!

closed 2020-07-08 by tedsalmon