tedsalmon/BlueBus #23: (Feature request) E39 non-nav enhancements

tedlee1688, 2020-07-07:
After getting the BlueBus wired into my E39 (2003 540i) noticed a couple of small improvements that can be made for Non-Nav E39 support:

- The BlueBus outputs into the spot on the MID where the clock and OBC data gets output and seems like it's competing with those functions for display space. 
[image]
On occasion the display flashes and then goes back to showing the clock and then flashes again and shows all the BlueBus text
- Is there a way to pass caller ID or address book through the HFP?  I upgraded from a dead Bluetooth TCU which had support for both.
 - Completely left field, but would be cool if the BlusBus could utilize the SOS/Wrench buttons as quick dials

tedsalmon, 2020-07-08:
@tedlee1688,

There are improvements in the pipeline that will address some of these requests. Lets break it down

>The BlueBus outputs into the spot on the MID where the clock and OBC data gets output and seems like it's competing with those functions for display space.

This is an unfortunate side-effect of DSP-enabled MIDs. On my non-BC, non-DSP MID, the BlueBus is able to use the entire screen space. Really, I think I'm going to hi-jack "TEL" display mode so I'm not competing for screen space. The implications of this are that the MODE, FM and AM buttons probably won't work anymore, which is why this is still swirling in my brain and not in the codebase.

>Is there a way to pass caller ID or address book through the HFP? I upgraded from a dead Bluetooth TCU which had support for both.

Yes, but this is far more complex than you could imagine. I only get a phone number passed via the HFP connection right now, so I have to pull your entire phone book and cross reference it with that phone number to see if there's a friendly name to display. For the phone book, I again have to pull the whole thing and store it on the EEPROM, then write the code to iterate over that and display it on the screen based on user input. This is way on the bottom of the list in terms of features, but it is planned.

>Completely left field, but would be cool if the BlusBus could utilize the SOS/Wrench buttons as quick dials

Possible, but I believe there are side-effects to using these buttons in the car. At least my Nav-equipped E46 goes into a tizzy if you use the SOS button.

Thanks!
-Ted

tedsalmon, 2020-09-28:
@tedlee1688,

This was resolved in 48396c9d79bce5fa62f93acaa161aeb44b728ad1

Both the UI issues and the Caller ID. Here's a binary: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_15_3.hex

Thanks!
-Ted

closed 2020-09-28 by tedsalmon