tedsalmon/BlueBus #26: Multifunction Wheel Voice button on Z4

zsoden, 2020-07-29:
My 2006 E85 Z4M can play and pause tracks using the voice button on the steering wheel but it doesn't launch Google assistant on a long press. It seems that the Z4M wheel does not send the IBusMFLButtonVoiceHold command. Here is a packet capture while holding the button:

```
198472] DEBUG: IBus: RX[6]: 50 04 B0 3B 80 5F 
[198477] DEBUG: MFL OFF
[198815] DEBUG: IBus: RX[8]: A4 06 BF 70 03 00 1A 74 
[199407] DEBUG: IBus: RX[6]: 68 04 FF 02 00 91 
[199857] DEBUG: IBus: RX[6]: 50 04 B0 3B A0 7F 
[199864] DEBUG: MFL PRESS
```

Pressing and holding the phone button on the head unit does trigger Google assistant, just not the talk button on the steering wheel. It would be good to try out on another Z4 to confirm that it's not just a fault with my steering wheel unit and come to think of it I'm sure that the stock TCU module didn't accept voice commands on a long press either (that module was faulty and removed long ago).

Perhaps a solution is to integrate a timer? If more than 2 seconds elapses with no IBusMFLButtonVoiceHold or IBusMFLButtonVoiceRelease event then trigger BC127CommandToggleVR() and ignore the next release event.

tedsalmon, 2020-09-28:
I've asked over on ZPost for someone to confirm :)

tedsalmon, 2021-04-22:
@zsoden,

I believe your MFL is faulty as I have verified with a local Z4M customer that he can summon the assistant via the MFL :/.

Thanks!
-Ted

closed 2021-04-22 by tedsalmon

zsoden, 2021-04-27:
Roger that, thanks for the confirmation Ted. I've learned to live with it by just using the phone button on the head unit instead :)