tedsalmon/BlueBus #30: Linker issue - application

ShonkyCH, 2020-09-04:
Hi 

I'm in the process of building one from your design. I have the parts and PCBs on the way, but decided to make sure I could also compile the firmware since there's a fair chance I can contribute there.

I'm running MPLAB X IDE v 5.20 and XC16 v1.60 (free version) on Windows. Of significance it appears the project originally/still uses XC v1.35 for the bootloader and XC v1.36 for the application. Not sure if that might cause an issue.

Bootloader appears to compile just fine but the application gives errors about some UART interrupt sections/symbols and a couple of others. I'm not experienced with PICs or XC16 in particular so thought it might be easier to just ask since it mostly works.

This is fairly clean installation of MPLAB IDE X and XC16. It did give an intial warning about the toolchain not being specified in the project but seemed to figure it out and the bootloader compiled just fine:

```
warning: Configuration "bootloader" builds with "XC16", but indicates no toolchain directory.
info: Configuration "bootloader" will build with toolchain "XC16" at "C:\Program Files\Microchip\xc16\v1.60\bin".
warning: Configuration "application" builds with "XC16", but indicates no toolchain directory.
info: Configuration "application" will build with toolchain "XC16" at "C:\Program Files\Microchip\xc16\v1.60\bin".
```

application build output

```
CLEAN SUCCESSFUL (total time: 30ms)
make -f nbproject/Makefile-application.mk SUBPROJECTS= .build-conf
make[1]: Entering directory 'C:/Users/ShonkyCH/Projects/BlueBus/repo/firmware/application'
make  -f nbproject/Makefile-application.mk dist/application/production/application.production.hex
make[2]: Entering directory 'C:/Users/ShonkyCH/Projects/BlueBus/repo/firmware/application'
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/bc127.c  -o build/application/production/lib/bc127.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/bc127.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/eeprom.c  -o build/application/production/lib/eeprom.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/eeprom.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/i2c.c  -o build/application/production/lib/i2c.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/i2c.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/char_queue.c  -o build/application/production/lib/char_queue.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/char_queue.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/event.c  -o build/application/production/lib/event.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/event.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/config.c  -o build/application/production/lib/config.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/config.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/ibus.c  -o build/application/production/lib/ibus.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/ibus.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/log.c  -o build/application/production/lib/log.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/log.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/utils.c  -o build/application/production/lib/utils.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/utils.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/pcm51xx.c  -o build/application/production/lib/pcm51xx.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/pcm51xx.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/uart.c  -o build/application/production/lib/uart.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/uart.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/sfr_setters.s  -o build/application/production/lib/sfr_setters.o  -c -mcpu=24FJ1024GA606  -omf=elf -DXPRJ_application=application  -legacy-libc  -Wa,-MD,"build/application/production/lib/sfr_setters.o.d",--defsym=__MPLAB_BUILD=1,-g,--no-relax
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/timer.c  -o build/application/production/lib/timer.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/timer.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   lib/wm88xx.c  -o build/application/production/lib/wm88xx.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/lib/wm88xx.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   ui/cli.c  -o build/application/production/ui/cli.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/ui/cli.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   ui/bmbt.c  -o build/application/production/ui/bmbt.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/ui/bmbt.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   upgrade.c  -o build/application/production/upgrade.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/upgrade.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   ui/mid.c  -o build/application/production/ui/mid.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/ui/mid.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   ui/cd53.c  -o build/application/production/ui/cd53.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/ui/cd53.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   main.c  -o build/application/production/main.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/main.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   handler.c  -o build/application/production/handler.o  -c -mcpu=24FJ1024GA606  -MMD -MF "build/application/production/handler.o.d"      -mno-eds-warn  -g -omf=elf -DXPRJ_application=application  -legacy-libc    -O1 -msmart-io=1 -Wall -msfr-warn=off  
[... cut ...]
nbproject/Makefile-impl.mk:39: recipe for target '.build-impl' failed
 Link Error: Could not allocate section '.aivt._AltMathError' at 0x180c
 Link Error: Could not allocate section '.aivt._AltNVMError' at 0x1808
 Link Error: Could not allocate program memory

Info: Loading file: ./application.gld
make[2]: *** [dist/application/production/application.production.hex] Error 255
make[1]: *** [.build-conf] Error 2
make: *** [.build-impl] Error 2

BUILD FAILED (exit value 2, total time: 5s)
```

Any idea?

Thanks

tedsalmon, 2020-09-04:
@ShonkyCH,

It appears Microchip broke something in `xc16` 1.60. Please download xc16 v1.50 [from here](https://www.microchip.com/development-tools/pic-and-dspic-downloads-archive).

I can't get my application code linked using the the default linker script when using v1.60 because it's having trouble allocating the memory for my alternate interrupt vector table. I'll wait for the next `xc16` release to see if the issue has been resolved.

Thanks!
-Ted

tedsalmon, 2020-09-09:
@ShonkyCH - I got things to work with 1.60 in d1110d1f1595e510ccaf1c90efe71a7a57408cef, just FYI :)

ShonkyCH, 2020-09-09:
Thanks 👍 

I just received the rest of my parts today, so will be building my unit tonight.

ShonkyCH, 2020-09-09:
Now the bootloader doesn't build for me :) (application builds but I can't get it to run in the debugger for some reason just yet)

```
"C:\Program Files\Microchip\xc16\v1.60\bin\xc16-gcc.exe"   -o dist/bootloader/production/bootloader.production.elf  build/bootloader/production/lib/eeprom.o build/bootloader/production/lib/flash.o build/bootloader/production/lib/protocol.o build/bootloader/production/lib/sfr_setters.o build/bootloader/production/lib/timer.o build/bootloader/production/lib/uart.o build/bootloader/production/lib/utils.o build/bootloader/production/main.o      -mcpu=24FJ1024GA606        -omf=elf -DXPRJ_bootloader=bootloader  -legacy-libc    -Wl,,,--defsym=__MPLAB_BUILD=1,,--script="bootloader.gld",--memorysummary,dist/bootloader/production/memoryfile.xml  -mdfp="C:/Program Files (x86)/Microchip/MPLABX/v5.40/packs/Microchip/PIC24F-GA-GB_DFP/1.3.109/xc16" 
c:\program files\microchip\xc16\v1.60\bin\bin\..\bin/elf-ld.exe: Warning: Invalid FBSLIM setting.
  calculated end address of 0x002000   is not within bounds:
  0x000100   to 0x0019fe   for 'program' region
```

ShonkyCH, 2020-09-09:
Maybe related to the application not working but seems to have broken with this update:

```
Warning: <redacted path>/firmware/application/dist/application/production/application.production.hex contains code that is located at addresses that do not exist on the PIC24FJ1024GA606.
```

tedsalmon, 2020-09-09:
@ShonkyCH,

Pull down 4631fdc839bc0291c5ff7807c89c8be0df90359c please :smile: 
The FBSLIM warning is inconsequential. It;s complaining that the AIVT is outside the bounds of the bootloader memory space, which is the case because the application contains the AIVT.

-Ted

ShonkyCH, 2020-09-09:
Yeah that seems to have fixed it. I read the the *red* warning as an error. And yes the application does run properly now too. Thanks.

closed 2020-09-09 by tedsalmon