tedsalmon/BlueBus #32: Incorrect behavior of comfort blink on E39 (non-nav MID)

vgergiev, 2020-09-08:
When Blink Option is enabled (options 3 or 5) it can glitch when user is quickly switching turn lights. Issue is not reproducing when Blink Option = 1.

Steps:
1. Turn on left blink
2. Immediately, while first blink is still on switch to the right blink.
3. Immediately, while first blink of right turn is on turn off blinking.
4. Turn on left blink

Actual result: 
Left turn is blinking even when turn stick is in the middle position (at the rest).
Enabling left turn is not affecting this behavior. Only right turn can disable this blinking.

Expected behavior:
Turn signals are working as designed :D

Video 1: https://drive.google.com/file/d/11C5rW3stquAbVVfyfi1S8gaNT3GlSkAX/view?usp=sharing
Video 2: https://drive.google.com/file/d/1osLPaM-Q1jd5DugnLn4fnRuG27WQ9Qu_/view?usp=sharing

0x200h, 2020-09-08:
Also had this problem happen to me last week - though I thought it was the same cause as a previous problem I had when the gold plate inside the stalk had been scraped between the two contacts causing conductivity - it went away after I indicated right - which didn't happen with the gold plate problem

harryberlin, 2020-09-08:
handling of turnsignal is a bit tricky. one point, you should consider the hazard lights too.

tedsalmon, 2020-09-09:
There is a lot of logic lacking around how turn signals work. The bug here is caused by the fact that the second bulb activation places the system in "Off" mode as it receives another diagnostics ACK from the LCM (which is what we expect when terminating diagnostics). The code needs to be abstracted and there needs to be better handling of the diagnostics ACK message.

tedsalmon, 2020-09-28:
@vgergiev / @0x200h,

This has been resolved in commit 9824bc4c947f2c56728b8c4f0d3614bfc3597a77 -- Here is a usable binary: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_15_3.hex

Many thanks to @piersholt for refactoring this functionality and making it way more usable. Please see PR https://github.com/tedsalmon/BlueBus/pull/37 for more :)

Thanks!
-Ted

closed 2020-09-28 by tedsalmon

piersholt, 2020-09-29:
Correction. I made it _awesome_!