tedsalmon/BlueBus #33: BC127 Firmware Versions

ShonkyCH, 2020-09-09:
I have built my own unit. Worked first time although I did test as I built it :) However I seem to have an issue with the BC127 module that I purchased. Possibly I have the wrong version or firmware. I bought the AptX version from Digikey with the Sierra Wireless part number "BC127-HD_1103709"

Firstly it defaults to 9600 baud, but BlueBus talks to it at 115200 looking the TX/RX to the module. I modified BlueBus to talk at 9600 and I can capture the comms on a logic analyser. BlueBus seems like it might be getting a bit confused as it sends the "STATUS" command before VERSION has finished transmitting presumably as it's ~10 times slower at 9600.

VERSION returns: (hand copied from logic analyser so might not be 100% correct)
```
BlueCreation Copyright 2017
Melody Audio V6.1.5 HD
Build: 1496941831
Bluetooth address 20FABBxxyyzz
Profiles: A2DP AVRCP HFP SPP PBAP MAP
Codecs: SBC APTX-HD
OK
```

STATUS returns
```
STATE CONNECTED[0] CONNECTABLE[ON] DISCOVERABLE[ON] BLE[IDLE]
OK
```

Have I got the wrong version here? Or is it likely fixable if I can configure the module to talk at 115200 by default, which I presume shouldn't be too hard? My understanding is they are largely identical in terms of operation. However I see 6.1.1, 6.1.2, 6.1.5 and 7.1/7.2/7.3 versions all listed on the SW website.

ShonkyCH, 2020-09-09:
115200 doesn't seem to particularly help.

ShonkyCH, 2020-09-10:
So looks like one of the issues is the v6 Melody firmware doesn't support the BT_STATE command (returns ERROR 0x0012 which is an unsupported command).

Is there any point trying to get v6 to work, or should I just try and get the module upgraded to Melody v7?

ShonkyCH, 2020-09-10:
Making progress. Stop me now if you know v6 is a not the way to go. :)

BT_STATE replaced with two commands CONNECTABLE ON/OFF an DISCOVERABLE ON/OFF seems to resolve most issues. It doesn't like one of the OFF commands in a certain mode but doesn't seem to be a big issue.

Metadata doesn't seem to be working. Unit just returns OK. I'm pretty sure my phone is correctly sending it (well works in an F31)

tedsalmon, 2020-09-10:
@ShonkyCH,

You MUST upgrade the BC127 to Melody 7.3 -- You can continue changing the firmware so that it supports the 6.1.5 API, but you're going to put yourself in a position where you won't be able to upgrade the BlueBus. Beyond that, the BC127 is way more stable after Melody 7.x.

Use the bootloader tool to place the BlueBus in `--btmode` and then use the Melody Updater to upgrade to 7.3.

Thanks!
-Te

ShonkyCH, 2020-09-10:
Thank you again for your help. Upgraded to 7.3 HD and it's all working quite well now.

closed 2020-09-10 by ShonkyCH