tedsalmon/BlueBus #36: When skipping track, short drop out in previous track

ShonkyCH, 2020-09-10:
Not sure if this is related to my homebuilt unit

Melody 7.3 HD firmware.
Latest github application firmware.
E46 i.e. analog audio
Galaxy S8+ (I know this is noted as a potential issue but works fine otherwise so far for me - no dropouts or pauses apart from this)

When skipping, the sound briefly drops out on the previous song. 
Then it plays a bit more (maybe 0.5s) of the previous song and then switches to the next song.

I don't think this is normal as it's quite disconcerting/annoying. If the little drop out was slightly later as the tracks changed, I think that would be better. Is that what the dropout is for but somehow the timing is off?

harryberlin, 2020-09-11:
that's the radio mute by track changing. you can fix that, if you set back to play, when next track begins to play. if you get this state or callback.

ShonkyCH, 2020-09-11:
Thanks for the suggestion. So is that something on the BMW side doing the muting?

I don't quite follow your "set back to play, when next track begins to play" idea? Wait for the next track to start, wait for the track to start playing and *then* send a "play message" on to the ibus?

harryberlin, 2020-09-11:
yes, bmw side. the radio mutes during track changing.
check the cd changer emulation.

tedsalmon, 2020-09-11:
The radio doesn't mute out the audio if your CD Changer emulation ignores the `0x02` command via `0x38` from the `RAD`. I just send back a "Playing" response to the radio and it is none the wiser.

@ShonkyCH,

What you're experiencing is a bug in the BC127 BT AVRCP stack. There's unfortunately nothing I can do about this :(.

Thanks!
-Ted

ShonkyCH, 2020-09-11:
Could you temporarily mute when receiving the skip push button, then send the skip AVRCP and then unmute? It is kinda annoying

tedsalmon, 2020-09-11:
@ShonkyCH,

I _could_, but that might clip the beginning of the next track depending on the timing. I will play around with a ~250ms mute of the DAC.

Thanks!
-Ted

ShonkyCH, 2020-09-11:
Yeah I realise it might be hard to time it just right so was just throwing it out there as an idea. Thanks for having a go.

tedsalmon, 2022-10-05:
SO! I recently realized that this is actually the radio cutting out audio after you press the next / previous track buttons! I don't know why it does this, but it does. If you use the phone controls, this issue doesn't occur.

So, `cantfix` :(

closed 2022-10-05 by tedsalmon