tedsalmon/BlueBus #41: Previous and Next buttons on MID is not working

vgergiev, 2020-10-13:
After update with moving to TEL UI buttons, which are on MID, are not switching the tracks.
The buttons on the wheel are working well, without any issues.

It seems that BlueBus is not listening for commands issues by these buttons.

[image]
(Photo from internet)

Probably it can affect those guys who don't have media buttons on the wheel.

piersholt, 2020-10-15:
If a button has multiple use cases the BMBT will address a specific device rather than broadcasting a button message. I gather the MID is similar in this respect.

I'd guess the button message handler needs to be updated accordingly. I'll check it later today.

tedsalmon, 2020-10-15:
@vgergiev,

This has been fixed in the newest build (code hasn't been pushed up because... reasons?) https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_15_9.hex

When I switched to the TEL UI handler, the RAD stopped sending the CDC the "Next Track" and "Previous Track" commands. I had not been listening to the physical button press messages before this change, so I had to add a new event.

Thanks!
-Ted

closed 2020-10-15 by tedsalmon