tedsalmon/BlueBus #43: Video Modul 16:9 Screen override

dertimaushh, 2020-10-22:
Hi Ted, 

i've realized that Bluebus overrides the signal of my TV Modul.  I'd like to use the TV Modul to connect it to an CarPlay adapter and use the Bluebus Bluetooth connection for Voice Controls and Audio. This would be the perfect 2021 tech update while maintaining the 100% OEM look.  

In reverse it works fine and my camera overrides the screen as designed. 

The setup in my 2001 E38: BM54 / Bluebus / 16:9  / MK4  / TV-Modul / Reverse Camera 

Any thoughts on this?

Best regards,
Tim

tedsalmon, 2020-10-23:
Tim,

Would it be possible for you to provide a log (https://github.com/tedsalmon/BlueBus/wiki/Gathering-a-Log) of the BlueBus overriding the TV Module? It would also be useful to see a video of what you're describing.

I suspect I'm simply not listening to a specific I-Bus message notifying the system that the TV Module is active. We didn't get the TV module here in the U.S., so it's not something I'm very familiar with.

Thanks!
-Ted

dertimaushh, 2020-10-26:
Hi Ted,

Here is the log:
[bluebus.txt](https://github.com/tedsalmon/BlueBus/files/5441301/bluebus.txt)

On video:
Without Bluebus:
https://youtu.be/c_i9hbMeLc8
With Bluebus:
https://youtu.be/KlGnF2iGF6c

Many thanks,
Tim

harryberlin, 2020-10-26:
GT tells the Radio
3B 05 68 4E 01 00 19: TV MODE ON
3B 05 68 4E 00 00 18: TV MODE OFF

dertimaushh, 2020-10-27:
That's great - i can see 3B 05 68 4E 01 00 19 twice in the Log - because i've tried it twice :-)

piersholt, 2020-10-27:
@dertimaushh 

Sounds like my kind of crazy! 😄  How on earth do you get a CarPlay adapter output a suitable video signal?

dertimaushh, 2020-10-27:
The are some Apple carplay interfaces out there with AV / RGB Output which i've seen working on the 16:9.  But 1st Bluebus should stop overwriting the screen in TV Mode :-) 

E.g. SUNE TECH Wirless Apple Carplay & Android Auto Box with USB player / Music / BT

harryberlin, 2020-10-27:
> That's great - i can see 3B 05 68 4E 01 00 19 twice in the Log - because i've tried it twice :-)

you can't do nothing. just wait for update.
that's a bug of bluebus. the messages are required to solve the problem.

dertimaushh, 2020-10-27:
> you can't do nothing. just wait for update.
> that's a bug of bluebus. the messages are required to solve the problem.

I know - I was just excited about the fact that my logging worked.

piersholt, 2020-10-27:
@dertimaushh 

> The are some Apple carplay interfaces out there with AV / RGB Output which i've seen working on the 16:9. 

Ahah, who knew!?

> I know - I was just excited about the fact that my logging worked.

Thanks for taking the time to do the logging, not to mention the videos. It's a huge help.

> But 1st Bluebus should stop overwriting the screen in TV Mode :-)

It's in the pipeline 😊

tedsalmon, 2020-11-03:
@dertimaushh,

Please try this firmware: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_16_2.hex

Thanks!
-Ted

dertimaushh, 2020-11-03:
Almost same result: https://youtu.be/ddkMxfc9Cy4

But while driving the TV Waring while driving message stayed occasionally when I hit the view toogle button several times. 

So some progress :-)

Best regards,
Tim

tedsalmon, 2020-11-09:
@dertimaushh 

Try this: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_16_5.hex

Thanks!
-Ted

dertimaushh, 2020-11-09:
[image]

It's working!!!! Many thanks.

Next question - not Bluebus related: How can I keep the Bluetooth Audio connection alive when switching to TV Mode?!

tedsalmon, 2020-11-09:
@dertimaushh,

I don't think that's possible. The `3B 05 68 4E 01 00 19` message tells the radio to switch audio inputs as well as "stopping" the CD Changer. It should resume when you exit TV mode, though.

Thanks!
-Ted

dertimaushh, 2020-11-09:
Yes it resumes when I exit TV mode. 

Guess it's not possible havin TV mode without that bit (3B 05 68 4E 01 00 19)?

tedsalmon, 2020-11-09:
@dertimaushh 

Issuing the inverse of `4E` (`0x00` rather than `0x01) will result in the radio writing to the GT again, and therefore booting you from TV mode anyways (similarly to the initial bug reported here). If you had the DSP Amplifier, I could trick the DSP into using the S/PDIF input rather than analog, but based on your YT videos, this vehicle doesn't have DSP :(.

-Ted

dertimaushh, 2020-11-09:
No, no Dsp. That's an other upgrade  😄

I think I'll have to work with the GPIO-Pin 7 of TV-Module as the Raspberry does.

tedsalmon, 2022-10-05:
This has been resolved for some time now :)

closed 2022-10-05 by tedsalmon