tedsalmon/BlueBus #45: Set Vehicle Time

tabor-a, 2020-10-26:
As discussed on NA M3 Forum, hack something into the BlueBus's firmware so that every time the car turns on, the BlueBus optionally gets the time off of the phone it connects to and sets the car's time.

tedsalmon, 2020-10-26:
@tabor-a -

This is very much outside the scope of what's possible right now. There's no inherent way of pulling the date and time via any of the Bluetooth profiles available. It would be possible to do this if there were a BlueBus app that was specifically programmed to push the date and time to the BlueBus in a way it could interpret.

That's not to say I don't want to do this, because I do, but this is near the bottom of the features to implement because of the barrier to entry created by having to make an app.

Thanks!
-Ted

piersholt, 2020-10-26:
@tabor-a 

To follow on from @tedsalmon, there's also an inherent constraint with respect to setting the time.

Clusters coded for BMBT will [accept input](https://github.com/piersholt/wilhelm-docs/blob/master/gt/40.md#time) from GT for multiple parameters including time. However, clusters coded for MID have no need for input, thus time cannot be set.

That's not to say it's not possible, but it's a case of diminishing returns as Ted said.

RosieE39, 2020-10-27:
You could implement this on cars with navigation by requesting the GPS time from the NAV computer.

harryberlin, 2020-10-27:
the point is, he don't get the time from phone.

piersholt, 2020-10-27:
@RosieE39 the navigation computer does indeed [broadcast GPS Time](https://github.com/piersholt/wilhelm-docs/blob/master/nav/1f.md), however it's UTC.

The timezone can _technically_ be deduced from [GPS coordinates](https://github.com/piersholt/wilhelm-docs/blob/master/nav/a2.md), but I'm unaware of a C library that's not an order of magnitude larger than the flash memory capacity 😋

Ted and I also looked at time from RDS, but there's only a handful of regions this is supported. That's also the case with RCC.

piersholt, 2020-10-27:
@harryberlin

> you can code the ike to receive the gps time for clock.

Code it and see what happens 😊

RosieE39, 2020-10-27:
I'm unaware of a C library that's not an order of magnitude larger than the flash memory capacity 😋

What about if you had a manual setting for Timezone (+/- HRS UTC)?

harryberlin, 2020-10-27:
> @harryberlin
> 
> > you can code the ike to receive the gps time for clock.
> 
> Code it and see what happens 😊

don't need it. have rcc from e38 😜
why do you need this feature? is your clock working incorrect?

tabor-a, 2020-10-27:
@harryberlin  - why do you need this feature? is your clock working incorrect?
Like many others, I work on my cars frequently and disconnect the battery.  It would just be nice to have the clock reset automatically.  Earlier, I was also having issues with the battery getting drained when the E46 was not used for a few days, which now seem to be resolved.

tedsalmon, 2020-10-27:
> > I'm unaware of a C library that's not an order of magnitude larger than the flash memory capacity yum
> 
> What about if you had a manual setting for Timezone (+/- HRS UTC)?

This is a pain in the ass for those of us who live in timezones that make use of DST.

> @harryberlin - why do you need this feature? is your clock working incorrect?
> Like many others, I work on my cars frequently and disconnect the battery. It would just be nice to have the clock reset [...]

To add to this, many E46 users without Navigation are unable to set the clock at all because the IKE adjustment "knob" often breaks.

I've tried a couple things since this issue was opened:

* `AT+CLLK?` via HFP. This is not supported as Android only supports SET style Hayes commands

* BLE `Current Time Service (0x1805)`. I'm a bit hazy on BLE, but it seems this would ALSO require a bespoke app.

-Ted

tabor-a, 2020-10-27:
The BLE Current Time Service was what I had originally thought of...
https://os.mbed.com/users/ohneta/code/BLE_CurrentTimeService/
or possibly something like
https://github.com/prasantmisra/cheep-sync

tedsalmon, 2020-10-28:
For what it's worth, I did some tinkering with the nRF connect app (which supports the current time service):
```
# set gatt
[43691] DEBUG: BT: Send Command 'BLE_GET_SERV 24'
OK
# [43710] DEBUG: BT: PENDING
[43785] DEBUG: BT: BLE_SERV 24 U16 1801 0001 0003
[43790] DEBUG: BT: BLE_SERV 24 U16 1800 0014 001A
[43795] DEBUG: BT: BLE_SERV 24 U16 1805 0028 FFFF
[43799] DEBUG: BT: OK

# set ble_read 002A
[52199] DEBUG: BT: Send Command 'BLE_READ 24 002A'
OK
[52217] DEBUG: BT: PENDING
[52322] DEBUG: BT: BLE_READ_RES 24 002A A E4070A1C0B041D035700
[52326] DEBUG: BT: OK
```

The format appears to be big-endian --
```
0xE407 = 2020
0x0A = 10
0x1C = 28
0x0B = 11
0x04 = 04
0x1D = 29
```

This will still require an App, but it's nice to know how it'll work.

-Ted

ShonkyCH, 2020-10-28:
0xE407 = 2020 is little endian :)

tedsalmon, 2020-10-28:
> E407 = 2020 is little endian :)

Ah yes, duh! I got confused by the ordering of the entire value.