tedsalmon/BlueBus #47: E39 MID: Display does not clear correctly

E39NLA, 2020-11-02:
E39 with DSP and MID display. Upon hitting the power button, the lower band of the display does not clear correctly. Upper band may also continue to print "Bluetooth" if accessing bluebus through the AUDIO button (not so when thru TEL button). 

[image]

tedsalmon, 2020-11-02:
@E39NLA ,

What firmware are you running? The "audio" UI is expected to be a bit wonky, but this recently got loads better when I fixed the response to `0x02` from the emulated TEL system (which cause a MID redraw request every 10 seconds).

Thanks!
-Ted

E39NLA, 2020-11-02:
Hey ted! I'm not totally sure because it was one of the incremental updates from the TEL button enhancement thread. Where can I find the master updated firmware? (I know this is a silly question, but all I can figure out how to download form the "main" project page on GitHub is the source code - the wiki wasn't helpful on this).

tedsalmon, 2020-11-02:
@E39NLA,

https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_16.hex

Thanks!
-Ted

E39NLA, 2020-11-12:
Just tested with that firmware. Still does it.

tedsalmon, 2020-11-12:
@E39NLA 

To be clear, the issue that you're experiencing is that upon boot, the MID doesn't get consistently written to. right?

Thanks!
-Ted

E39NLA, 2020-11-12:
I should clarify - this happens when I am powering OFF, not on. The upper band clears out correctly and leaves just the clock. The lower band continues to be displayed.

No issues on boot.

tedsalmon, 2020-11-12:
Can I please get a log? https://github.com/tedsalmon/BlueBus/wiki/Gathering-a-Log :)

Thanks!
-Ted

rsporsche, 2021-04-09:
I have a MID interface on my E39 and I updated to 1.1.16 yesterday and I can reproduce this behaviour.

I've attached a log that I captured while I pressed various buttons including settings, cycling audio modes, power on/off, BC/Clock, etc. If you need a log with a specific set of actions let me know and I will get it. I can also make a video of the behaviour if it helps.

[[Com COM3] (2021-04-08_172253).log](https://github.com/tedsalmon/BlueBus/files/6283160/Com.COM3.2021-04-08_172253.log)

tedsalmon, 2021-04-13:
@rsporsche 

Thanks! I will review this sometime this week ☺️

-Ted

rsporsche, 2021-04-13:
No problem, let me know if you need anything else, I was in a bit of a rush when I got the last log and I didn't take care to record which buttons I pressed.