tedsalmon/BlueBus #48: CD43 Support

E39NLA, 2020-11-02:
Enhancement request - are there any plans to support the CD43 head unit with bluebus? This would provide an obvious upgrade path to bluebus for E32/34/36 fans I would think. I believe this the only single-DIN style head unit out there with ibus support (the C33 and C43 cassette head units are "m-bus" I believe).

tedsalmon, 2020-11-02:
@E39NLA -

Thanks! The CD43 is already supported and I have a couple customers using a BlueBus in an E36 / E36 Z3.

Thanks!
-Ted

E39NLA, 2020-11-02:
Ah. Then I might have a technical glitch here. I have a CD43 in my E34 with a late E36 CD changer harness (that is, the 3pin ibus with 6pin audio connectors) spliced in to the radio harness. The ibus wire from the 3pin connector is run directly to Pin 7 at the back of the CD43. Pressing "CD" just gives me NO DISC. Bluebus shows the green indicator light for a few minutes then seems to power down on it's own. I've already verified all wiring is good (continuity OK).

Will note I am using a "generic" CD43 (i.e. the one Rover Group made under license from BMW) but everything I've read online assures me that the non-BMW CD43's support ibus. I have bench tested it by hooking it to a spare E38 MID I have and they do talk to each other.

tedsalmon, 2020-11-02:
@E39NLA 

Try the following command in the BlueBus terminal:

`set ign alwayson`

That should allow it to work :smile: 
-Ted

E39NLA, 2020-11-02:
Will give that a shot! Will that have any impact on bluebus use in my E39? Right now I only have one unit which I am intending to share between my two cars.

tedsalmon, 2020-11-02:
@E39NLA,

Yes, it will make it so the BlueBus connects and starts playing any time the E39 is woken up (unlocked, door open, etc). This isn't a problem with the E34 because the CD43 doesn't come to life without the key in the ignition.

Thanks!
-Ted

E39NLA, 2020-11-02:
Ah. So I take it that the bluebus is normally looking for a "wake" message from the ibus connection, and this configuration option tells it to ignore and turn on whenever 12V is applied.

tedsalmon, 2020-11-02:
@E39NLA,

The BlueBus expects the IKE/KOMBI to broadcast the ignition status, at which point the value determines the state of the BlueBus. Since the E31/2/4/6 aren't I/K-Bus cars, this data is unavailable. Setting `ign alwayson` tells the BlueBus to broadcast the ignition on status to itself as soon as it wakes up.

 See: https://github.com/tedsalmon/BlueBus/blob/master/firmware/application/handler.c#L238

-Ted

E39NLA, 2020-11-02:
Terminal is just giving me a "Command Not Found" error when inputting SET IGN ALWAYSON. Updated the firmware to 1.1.16

tedsalmon, 2020-11-02:
@E39NLA,

Oops, try: 

`set ign alwayson on`

Similarly, to toggle it back off:

`set ign alwayson off`

See: https://github.com/tedsalmon/BlueBus/blob/master/firmware/application/ui/cli.c#L472

Thanks!
-Ted

E39NLA, 2020-11-02:
Okay, getting closer. Accepted the command and when installed in my E34 it would pair to my phone. But the CD43 doesn't want to talk to it still. "NO DISC" like before.

tedsalmon, 2020-11-02:
@E39NLA.

Do you have a laptop you can plug into the unit while it's in the car? I would like to see if you're getting any I-Bus traffic when you issue the command: `set log ibus on`

Thanks!
-Ted

E39NLA, 2020-11-02:
Yes I'll give that a shot.

E39NLA, 2020-11-02:
Ok, got it all plugged into my laptop in the car. Entered the log ibus command.

tedsalmon, 2020-11-02:
> Ok, got it all plugged into my laptop in the car. Entered the log ibus command.

Can you run the command "get ibus"? It should return something...

E39NLA, 2020-11-02:
Get ibus just returns "OK"

tedsalmon, 2020-11-02:
> Get ibus just returns "OK"

You have an issue with the I-Bus, then.

Either theres a continuity issue, or you need to pull the I-Bus up to 12v through a 10k resistor.

Thanks!
-Ted

E39NLA, 2020-11-03:
Just checked the voltages at my ibus connector. Red (batt) is 11.99v, yellow (ibus or Pin 7) is 11.50v. Is the wire order correct at the connector?
[image]

tedsalmon, 2020-11-03:
@E39NLA 

That looks fine, and the voltages are good. Can you run "reboot" before you run `get ibus`?

Make sure the radio is on, too.

Thanks!
-Ted

E39NLA, 2020-11-03:
Same as before. Get ibus yields only "OK." Rebooted both with and without the radio on.

tedsalmon, 2020-11-03:
@E39NLA,

And as a sanity check, everything works well in the E39 still?

Do you have an OE CD Changer to test on the E34?

Thanks!
-Ted

E39NLA, 2020-11-03:
Sadly I tossed out my E39 cd changer after I got the bluebus (it never worked anyways). May be able to borrow one from a friend in a few days.

Bluebus still works fine in the E39, no issues.

tedsalmon, 2020-11-03:
> Sadly I tossed out my E39 cd changer after I got the bluebus (it never worked anyways). May be able to borrow one from a friend in a few days.
> 
> Bluebus still works fine in the E39, no issues.

With `set log ibus on`, when you issue the `get ibus` command, are you seeing the frame come back in the buffer? You should if the line is "good".

E39NLA, 2020-11-03:
Found this old cd changer simulator I used to use for aux input. Plugged it in and the CD43 would talk to it. Played off my phone just fine (3.5mm jack) with clear sound.

So I guess the ibus on the CD43 is working?

[image]
[image]
[image]

tedsalmon, 2020-11-03:
@E39NLA 

Are you sure that's a CD43? I don't know them well enough to tell from the images, but what I do see is that your box is an M-Bus emulator, rather than I-Bus, which suggests the radio speaks with the CDC over M-Bus and not I-Bus...

Thanks!
-Ted

E39NLA, 2020-11-03:
Positive. I know the label says M-bus but that I believe is referring to the other side of the box, which is the DIN-8 connector used for Alpine CD changers. If you google the model (ABWM35) you'll see it was designed for I-bus BWM's. This box works in my E39 too, so I know it's an I-bus device.

tedsalmon, 2020-11-03:
> Positive. I know the label says M-bus but that I believe is referring to the other side of the box, which is the DIN-8 connector used for Alpine CD changers. If you google the model (ABWM35) you'll see it was designed for I-bus BWM's. This box works in my E39 too, so I know it's an I-bus device.

Are you based in the US? The only thing I can think of is that the BlueBus doesn't have enough of a pull up to 12v on the IBus line, so I can make a patch cable with a built in pull up.

Apart from that, I've got nothing, especially considering the BlueBus works fine in your E39.

-Ted

mykekfz, 2020-11-03:
OP what is the part number on the CD43?

there are 2 version of the CD43 mBUS version and IBus version.

E39NLA, 2020-11-06:
PN is 7640263342.

Ted: yes, I'm in Northern NJ.

E39NLA, 2020-11-08:
As an added test: have a donor E39 CD changer that I plugged into the CD43. Worked without issue.

tedsalmon, 2020-11-09:
@E39NLA 

Do you happen to have a 10k ohm through-hole resistor laying around?

-Ted

E39NLA, 2020-11-11:
No, but I can certainly get my hands on one. How would I wire that in?

tedsalmon, 2020-11-11:
> No, but I can certainly get my hands on one. How would I wire that in?

Pop the top off the BlueBus and gently wrap the resistor legs around the center and lower pins of the three pin connector of the BlueBus. I would get an 1/8watt or lower resistor to ensure the legs are thin enough to wrap around the PTH pins.

Thanks!
-Ted

E39NLA, 2020-11-12:
Couldn't I do the same by bridging the leads on the 3pin connector (I assume from iBus to ground)? I would prefer to keep modifications to car side if I can do it that way.

tedsalmon, 2020-11-12:
> Couldn't I do the same by bridging the leads on the 3pin connector (I assume from iBus to ground)? I would prefer to keep modifications to car side if I can do it that way.

Sure. You're bridging 12+ to I-Bus, btw :). This is a temporary test, too, so don't solder anything.

E39NLA, 2020-11-26:
Finally found time to try this and it seems to have done the trick! But I'm only getting a basic CD changer readout on the CD43. Am I supposed to be seeing more?

I guess I'll solder in the resister permanently on the E34 side?

[image]

tedsalmon, 2020-11-26:
> Finally found time to try this and it seems to have done the trick! But I'm only getting a basic CD changer readout on the CD43. Am I supposed to be seeing more?
> 
> I guess I'll solder in the resister permanently on the E34 side?
> 
> [image]

Please don't solder anything! I can make you an adapter if you give me two or three weeks :).

The TH3122.4 is just a bit more sensitive to a weak pull-up on the I-Bus, it seems.

tedsalmon, 2020-11-26:
@E39NLA 

I forgot to add, you will need to issue these commands to get a UI on this car. It won't be set automatically because the BlueBus expects a "full" I-Bus car so it can query the Light Module for the VIN and then determine if it needs to rethink the UI.
```
set ui 1
reboot
```

Thanks!
-Ted

E39NLA, 2020-11-26:
Gotcha. Will that cause any issues swapping from the E34 to the E39?

Will shoot you an email 👍

E39NLA, 2020-11-28:
So I've got it installed with a permanent 10kOhm bridging 12V to Ibus. Working good so far! I do however have a few minor issues:

1. Noise/interference from right channel that increases with engine rpm.

2. After entering "set ui 1" I got track readout on the CD43 but the next track/prev track became buggy (had to hit it twice in quick suggestion for it to register). Hitting the radio preset buttons doesn't seem to do anything.

3. With TTY connected im getting a lot of [DEBUG] messages. Will post log. Is this normal?

E39NLA, 2020-11-28:
[bluebus.log](https://github.com/tedsalmon/BlueBus/files/5610192/bluebus.log)

Log here. I also noticed that the bluebus appears to be on all the time (green LED). E34's don't really got to sleep like the later cars, so is there any way to specify an auto shutdown (or perhaps a new CD43-type profile that includes that? Just a thought).

tedsalmon, 2020-11-28:
@E39NLA 

The BlueBus is set to go to sleep after 61 seconds with no I-Bus traffic. Since the E34 doesn't include ignition status or any other IKE traffic on the I-Bus, it is impossible to make the BlueBus sleep after some time without coding in the ability for it to shut off on you mid-drive.

I'll check the log a little later -- thanks for that!
-Ted

E39NLA, 2020-11-28:
Wouldn't it be possible to have bluebus go to sleep after 61 second of no phone paired/bluetooth communication?

LMK if you need more logs pulled with different circumstances/setup.

tedsalmon, 2020-11-29:
> Wouldn't it be possible to have bluebus go to sleep after 61 second of no phone paired/bluetooth communication?
> 
> LMK if you need more logs pulled with different circumstances/setup.

@E39NLA 

Sure, but I don't particularly like that idea as if there is still I-Bus traffic at the time, there's a good chance the unit won't be able to shut itself off as it is bus traffic triggered.

Can I get a log (enable I-Bus logging and sys logging, please) of you leaving the car off for two minutes?

I'd you're unfamiliar these are the logging commands you need to run:

```
set log sys on
set log ibus on
```

Thanks!
-Ted

blehn, 2022-08-28:
@E39NLA Did you ever get your BlueBus working reliably with the CD43?

tedsalmon, 2022-10-06:
@blehn I don't know about @E39NLA but I do have a few users running it on the CD43 successfully.

Closing this as I believe we should be golden.

-Ted

SecT0uch, 2024-01-23:
I have the same issue, my Bluebus won't switch off.

It seems there is still ibus traffic with the car off and key out:
```
# get pwroff
Auto-Power Off: On
OK
# set log ibus on
OK
# [450007] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
[471007] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
[492007] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
[513007] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
[534006] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
[555007] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
[576006] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
[597007] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
[618007] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
[639007] DEBUG: IBus: RX[6]: 18 04 FF 02 01 E0 [SELF]
set log ibus off
OK
```

Could it be related to the alarm system ?

Z3 coupé (E36)
BMW Business CD RDS 
Model: BP0273
Made by Blaupunkt

tedsalmon, 2024-01-23:
@SecT0uch

Try:

`set ign alwayson on`

SecT0uch, 2024-01-24:
Unfortunately, the Bluebus still stays on