tedsalmon/BlueBus #49: Current Firmware Release link on Wiki Firmware Upgrade page

tabor-a, 2020-11-09:
Add a link to the Firmware current release download *OR* the repository of all firmware releases on the Wiki Firmware Upgrade page in addition to the Firmware Update Tool.

tedsalmon, 2020-11-09:
@tabor-a 

Listing the software versions is on my radar already, but I'm not going to put it in the Wiki because hardly anyone comes to GitHub or understands how to navigate it. I'm just going to make a `/software` or something on the BlueBus website.

Thanks!
-Ted

closed 2020-11-09 by tedsalmon

tabor-a, 2020-11-09:
That would also do it!

tabor-a, 2020-11-10:
Thanks, Ted!  Software (Firmware) page looks great.  On the Installation page, there's a typo "Instalation" at the top...

tedsalmon, 2020-11-11:
Thanks, @tabor-a -- fixed!