tedsalmon/BlueBus #51: Comfort Blinkers cause dash brightness adjustment to stop

calmelb, 2020-11-12:
Whenever comfort blinkers are on (and only when those are specifically blinking) the dash brightness control and the headlight level adjustment control are no longer controllable for a period of time after the last blink. 

However if the blink is only once or its set to the second detent (so fixed on) then it functions as normal.

Running 1.1.16 on an E46 Pre-LCI with the Blaupunkt Head Unit (Car has no auto-headlights and no Xenons)

piersholt, 2020-11-12:
**_Challenge accepted._**

tedsalmon, 2020-11-12:
@calmelb,

For clarity, @piersholt previously did some _serious_ refactoring of the comfort blinkers code. In that, he moved away from terminating LM diagnostics to simply shutting down all bulbs activated by diagnostics in order to avoid throwing the LM into limbo  in certain use cases. Leaving the LM in diagnostics mode is what's causing this issue. It clears up after 15 seconds as that is the timeout for diagnostics on LSZs.

Thanks!
-Ted

piersholt, 2020-11-13:
It's somewhat obfuscated unless you dive into the light module object files, but it's the inputs that are being activated, or more specifically, the voltage for the various input switches. An example of the blinker switch voltages:

```
STAT SWITCH BLK SHORT ON GROUND   0-1v
STAT SWITCH BLK LEFT ON           1-2.5v
STAT SWITCH BLK RIGHT ON          2.5-4v
STAT SWITCH BLK SERENITY          4-5v
```

Whenever a switch voltage is set, any additional input from the "real" switch will cause the switch to end up in limbo as Ted pointed out. In order to prevent this, the switch voltage needs to be reset. In this case, the blinker switch voltage is set to `0`.

There's a catch that isn't apparent in the functionality - the voltage of any potentiometers is transposed into the activation command. Given there's no "off position" (or positions, at all for that matter), if the voltage doesn't match the "real" switch voltage, then any vehicle functions controlled by potentiometers will change accordingly - manual vertical headlight aim (HVAC), automatic headlight beam throw adjustment (LWR), instrument backlighting (58G) etc...

I didn't consider the potentiometers as they are "set and forget" for me.

_I'll tinker..._

calmelb, 2020-11-13:
@tedsalmon Thanks for the explanation! Love how complex this all is

@piersholt I only happened to note as only once it reset the instrument backlighting to the lowest level and then I realised I couldn't adjust. Is also a set and forget for me but I thought I would at least file a bug. Is very fascinating how it all works (as someone who has had an interest in electronics and now a gaining interest in how cars communicate over the -BUS) to be fair

piersholt, 2020-11-13:
@calmelb 

> I only happened to note as only once it reset the instrument backlighting to the lowest level and then I realised I couldn't adjust. Is also a set and forget for me but I thought I would at least file a bug. 

It's certainly helpful feedback. Thanks for being constructive ☺️

> Is very fascinating how it all works (as someone who has had an interest in electronics and now a gaining interest in how cars communicate over the -BUS) to be fair

That's how it began for me, and I'd guess, for Ted too!

piersholt, 2020-12-16:
I had a chance to test this yesterday, and I was able to replicate the behaviour.

I tinkered with a possible viable solution, but I'll do some further testing.

JananasCzech, 2023-02-27:
Hi guys, i think that a have same issue like @calmelb
I will test turn off comfort blinkers when i get my car from repair shop :D
Brightness start switching from max setting to min setting 6 months ago and i cannot find whats wrong.
I try change Licht module but nothing change. Thanks for thread ! :)
I will write update when i got a time.

piersholt, 2023-03-11:
@JananasCzech this is a _very_ specific issue. You'll only encounter this if you're manually adjusting the instrument backlighting via the dial within about 20 seconds of the comfort blinkers being active.

JananasCzech, 2023-03-23:
@piersholt ok thanks for the info,
How to correctly get from this issue? 
unplug battery in car? unplug blueBus?

I changed the setting for comfort blinkers to 8 and after a few days to 5 and then to 4 (originally from 3) and it seems that it is working fine right now.

so far i have seen this problem occur only once in the last 3 weeks. Anyway, an improvement.