tedsalmon/BlueBus #52: BMW Z4M Bugs - Autoplay - Settings - Aux - Clock

maupineda81, 2020-11-14:
Hello Ted, I have noted several issues with my BlueBus in my Z4M... ; let me list them below.

1. The settings menu is not persistent, the bluebus continues to return to the main page, to adjust settings you need to do it very quickly before it kicks you out. This means the settings need to be adjusted quickly and blindly

2. When autoplay is off, the bluebus constantly pauses musing, if enabled, you cannot pause it. Also, you can perceive audio coming up and down as the system pauses and then auto-plays again.

3. When connected to the phone and streaming musing, I use the AUX mode to reproduce musing, it is as if the phone stays in bluetooth mode and pased

4. When in bluetooth, the clock button does not display the time, even when metadata is off

5. When the car and key are turned off, the musing stops, but the radio display remains on, I have not timed for how long until it goes to sleep.

I captured some videos, but they are too large...

tedsalmon, 2020-11-14:
@maupineda81 

Issues 1 and 2 are fixed in this firmware: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_16_8.hex (Right Click > Save File As)

Issue 3 is intended operation. If you want to use the AUX input, I suggest turning off Bluetooth temporarily. If you change the output on the phone, the AVRCP profile may even be closed on the phone side, which will prevent the BlueBus from ensuring the phone isn't playing when it shouldn't be.

4. Can I get a video of this? You can upload it to YouTube, Dropbox, iCloud, etc.

5.What's the last thing displayed? If you hit a button on the radio, does it clear it?

Thanks!
-Ted

maupineda81, 2020-11-15:
Hello Ted, the new firmware did fix issues 1 and  2, with one little exception, if I click button 6 to turn pairing on / off, that menu is not persistent, the same issue as with the settings before.

About issue 4, I noted that is more of a radio behaviour, whenever the radio is in "phone" mode the audio setting buttons do not work, it seems that when the BB is displaying Meta Data, or just reading Bluetooth on the screen, the radio goes in what I call "phone mode", thus disabling the settings buttons. I have noticed that the way around it is... set the screen to read Bluetooth ILO of metadata, then click the mode button, and it takes you out of the phone mode back to CDC (changer) mode, then the time displays if the clock icon is enabled, or reads TR07-001 and remains there. To return to the Bluetooth / Metadata screen ("phone mode") you just press button two, this step can be finicky and may require more than one press of button 2 (ideally should be just one press). Interestingly, if the screen is displaying metadata, pressing the mode button to get off phone mode and into CDC mode if just does not work, so the dance step is press 2 untill "Bluetooth" is displayed, the press mode and it goes to the CDC screen. For more ease of understanding me, let me add some clarifiers below...

I call "phone mode" when the screen reads Bluetooth or displays metadata (the attached image is an example of this)
I call "CDC" mode when the radio screen reads TR07, or TR07-001

About item 5, please see attached image, at this point, the radio is off (you can tell by the clock icon above send/end scripts is not present (I always have clock display on)... it takes about a minute to go off. What I found is that if I turn the radio off before switching ignition to position 0, then the displays goes off immediately.

[[image]
](url)

tedsalmon, 2020-11-16:
@maupineda81 

>if I click button 6 to turn pairing on / off, that menu is not persistent
This is by design. It's only letting you know that you changed the pairing state :)

>it seems that when the BB is displaying Meta Data, or just reading Bluetooth on the screen, the radio goes in what I call "phone mode",
Yes, the only method of writing to the E46/R5x radios is to pretend to be the OE phone system writing call data to the screen. The Siemens-made versions of the CD53 equipped in later model E8x vehicles takes that protocol a bit to the extreme in that it adds additional UI doo-dads that didn't exist in the original Alpine-made and Blaupunkt-made variants.

>this step can be finicky and may require more than one press of button 2 (ideally should be just 1 press)
I'll check this out. :)

>mode button to get off phone mode and into CDC mode just does not work (does not do anything other than blinking the display)
Yep, no way around this. You MUST use the "2" button before attempting to change modes on these Siemens units.

>it takes about a minute for the display to go off. What I found is that if I turn the radio off before switching the ignition to position 0, then the display goes off immediately.
I don't think there's anything inherently bad about this. I think the radio just displays the last text sent from the "phone" until it goes into sleep mode (about 1 - 2 minutes after the doors are closed). The BlueBus goes to sleep at the same time.

I might be able to clear the screen on ignition off. I'll look into it, but I don't have a car with this radio, only a benchtop module.

>When in CDC mode, pressing button 2 will take you back to phone mode.
The "2" button should be used for both stopping the metadata writing AND reinstating it. "MODE" shouldn't be used at all.

>One last "glitch" I noted is that when musing is being streamed, and the display is in CDC mode, you need to press the mode button twice to jump to the next mode (AUX). one press reads no CD, the second press jumps you to AUX. It would be great to make it go to AUX with just one press.

"NO CD" is your radio reporting that your single CD slot has no CD in it, not the BlueBus ;). Regardless, I cannot externally manipulate a CD53 radio.

Thanks!
-Ted

maupineda81, 2020-11-16:
Thanks, Ted for all of the responses and help with my questions. I think this clears almost all the concerns I found. if you find a solution for clearing the screen off when shutting the car off let me know, is a minor thing but would help with that seamless experience as if it was OE :)

Let me share other things I noted that are irrelevant for my case, but maybe important to others with 100% stock cars.

I found that the blinkers settings don't do anything, I currently have my car coded to do five blinks, I tried using the BB to set it to three and one just to play with it and they did not work. For me this is not an issue as my car follows the setting given by the module... perhaps this would only work in a car that is set to a default of one blink.

Lastly, I did not see the setting to lock at speed, is that something that is not available for the Z4? I ask only for curiosity as I have the car coded to that too.

Overall the product is awesome and is just fine as is!!!!

piersholt, 2020-11-17:
@maupineda81 

> I found that the blinkers settings don't do anything, I currently have my car coded to do five blinks, I tried using the BB to set it to three and one just to play with it and they did not work. For [...]

The comfort blinker feature is for vehicles _without_ factory comfort blinkers. There's _no_ relationship with the factory implementation in any respect.

I'd highly recommend setting the blink count to `1` to disable the feature.

tedsalmon, 2020-11-17:
@maupineda81 

> if you find a solution for clearing the screen off when shutting the car off let me know, is a minor thing but would help with that seamless experience as if it was OE :)

I'll play around with this. It does seem like the code already exists, so I might need to tweak it a little bit for the Siemens CD53.

>Lastly, I did not see the setting to lock at speed, is that something that is not available for the Z4? I ask only for curiosity as I have the car coded to that too.

It's only not available because it hasn't made it to the UI. I don't know that it will work as it will treat your car like an E46...

>Overall the product is awesome and is just fine as is!!!!

Glad you're enjoying it! :)

-Ted

tedsalmon, 2020-11-25:
@maupineda81 

Can I please get a log of the car being turned off but still displaying "Bluetooth"? https://github.com/tedsalmon/BlueBus/wiki/Gathering-a-Log

Unfortunately, it seems I have code to handle this already, and I can't reproduce it as I have a Nav-based E46 :(

Thanks!
-Ted

maupineda81, 2020-11-25:
Hi Ted, please find attached the log I created. There are two sessions in the log, but they are short and should be easy to go by

First session ends at line event 91494, when the radio is playing and I just switch off from position 1 to 0. I kept logging until the unit goes down

The second session ends at event 84780, when I do the same, with the exception that I turned the radio off manually just beore turning the ignition from position 1 to 0. a few lines above you can see the pause instruction, I assume this is when the radio was turned off.

then after that, I don't understand much, but the number of events and instructions are different.

I hope the log is useful, as it was a bit of a PITA LOL...

tedsalmon, 2020-11-26:
@maupineda81 

The log helped a lot! With it, I was able to figure out that I can't clear the Siemens CD53 the same was as the Blaupunkt / Alpine variants.

Do I remember you saying that you had the factory TCU still installed?

Thanks!
-Ted

maupineda81, 2020-11-26:
^^ Yes, I have TCU and BB combo... Do you think that is related to the screen not clearing?

tedsalmon, 2020-11-26:
> ^^ Yes, I have TCU and BB combo... Do you think that is related to the screen not clearing?

Could I get a log of you using the phone features from FM mode? 😄

The TCU obviously has to clear the screen, so the key to fixing this issue lies in the traffic between the radio and TCU :)

Thanks!
-Ted

maupineda81, 2020-11-26:
Ted, it will take me a good two months :(.

I just put the car away and I am flying abroad for two months tomorrow... I will see if someone from the Z4 forum can help. will send an IM to the other user that knows how to log.

tedsalmon, 2020-11-26:
I have a factory TCU with Bluetooth out of my 2006 E46 M3. I will see if I can re-code it to work with the Siemens CD53 and also jerry-rid my MFL into play.

Thanks!
-Ted

bwright657, 2020-11-30:
Hey Ted,

@maupineda81 asked me to chime in here since I've got the non-nav CD53 and pulled my TCU entirely when I installed my BlueBus.  I see the same behavior with the screen.  If I'm I'm playing audio through the BlueBus and turn the car off, the audio cuts off instantly but the screen stays on for 60 seconds.  If the stereo is off or in another mode, the screen turns off instantly.  Let me know if you need me to pull any logs.

tedsalmon, 2020-12-01:
@bwright657 

Thanks for chiming in! Could I possibly bother you to reinstall the OE TCU, then log you using it through the radio display? I specifically want to see how it clears off the radio display when you dismiss it.

Thank you!
-Ted

tabor-a, 2020-12-01:
For what it's worth, I just verified that the E46/CD53/BB combo does shut off instantly when the car is shut off when playing, so sounds like the Z4 may need a separate UI?

maupineda81, 2020-12-01:
> @bwright657
> 
> Thanks for chiming in! Could I possibly bother you to reinstall the OE TCU, then log you using it through the radio display? I specifically want to see how it clears off the radio display when you dismiss it.
> 
> Thank you!
> -Ted

Hi Ted, if Brian cannot do this (I understand is a bit of a tear up), I can do the log once I get back to the US next year.