tedsalmon/BlueBus #53: Error with television, caller ID and coolant temp

E46er, 2020-11-16:
Hi Ted,
A couple of bugs:
I have updated to 1.1.16 and noticed that now 1: I cannot select television mode. -it jumps straight back to cdc/ Bluetooth screen. 
2: the caller ID has stopped displaying correctly. - it shows “00z” or similar random characters. 
3: I’ve never been able to get the coolant temp to display with any firmware version so far. 
4: also the phone call volume still seems to jump very high a significant amount when a call comes in. This has improved after the 1.1.16 update but still requires turning the volume down to a bearable level during the call, to then turn it up again when the music returns after the call has ended. 

my setup:
2004 E46 with Mk4 Nav, BM54 radio, Television module with DVB card mod, Hi-Fi amp. 

Thanks

dertimaushh, 2020-11-16:
This fixes the TV Mode issue:

https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_16_5.hex

see related thread: https://github.com/tedsalmon/BlueBus/issues/43

The call volume is ok for me since 1.1.16 - but to be fair i don't drive a lot these days. 

Best regards,
Tim

tedsalmon, 2020-11-17:
@E46er 

I would install https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_16_8.hex to solve the television issue, similar to what @dertimaushh suggested. I did just fix that for him :)

2. I'm aware of this but haven't had a chance to check into it.

3. Hmm. What does it display? Does the coolant temp gauge work? This works fine in my 2006 E46 M3.

4. 1.1.16.4+ further improves call volume handling. I suggest you run a "RESTORE" on the BlueBus to clear out the current volume offset :)

Thanks!
-Ted

E46er, 2020-11-17:
Great thanks - that update "1.1.16.8" has sorted the Television flip back issue, and has improved the music / phone call volume issue.!
I have done some digging with the no show of the coolant temp display, and found it was a setting in the Nav module. I Re-coded to the GT module the default setting by processing an empty Man file and it now displays. I believe the setting "show arrival time" is the culprit and it needs to be set to: nichts aktiv. - maybe useful for someone else that has been fluffing around in those settings and ended up with the same "glitch" I experienced!!

Cheers Ted, great work..

dertimaushh, 2020-11-17:
For me "show arrival time" is active and Temp is showing correct. Don't think this is the root cause.

piersholt, 2020-11-18:
> 3: I’ve never been able to get the coolant temp to display with any firmware version so far.

> I have done some digging with the no show of the coolant temp display, and found it was a setting in the Nav module. I Re-coded to the GT module the default setting by processing an empty Man file [...]

> For me "show arrival time" is active and Temp is showing correct. Don't think this is the root cause.

This is down to coding, but `WIDESCREEN_MODI` is the culprit. If GT (`GT.C04+`) is coded for what is _effectively_ a "non-NG" radio, then the [particular field](https://github.com/tedsalmon/BlueBus/blob/master/firmware/application/ui/bmbt.h#L25) that coolant is written to is not displayed.

E46er, 2020-11-18:
> This is down to coding, but `WIDESCREEN_MODI` is the culprit. If GT (`GT.C04+`) is coded for what is _effectively_ a "non-NG" radio, then the [particular [...]

Yes - you're right!  I didnt see that had changed also  (originally it was "wert 3", then after default load it went to "wert 1")
thanks for correcting that..

piersholt, 2020-11-18:
> Yes - you're right! I didnt see that had changed also (originally it was "wert 3", then after default load it went to "wert 1")
thanks for correcting that..

Happy to be able to clarify. My esoteric knowledge is finally paying dividends!

tedsalmon, 2020-11-19:
@E46er,

Please ensure this version fixes your Caller ID issues: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_17_beta.hex

Thanks!
-Ted

E46er, 2020-11-19:
> @E46er,
> 
> Please ensure this version fixes your Caller ID issues: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_17_beta.hex
> 
> Thanks!
> -Ted

Hi Ted,
Tried it for 10mins or so and ran though testing various operations etc
Caller ID working.
Language selection working.
I had 2x incidents where the screen changed to the music dashboard whilst on a phone call. - this caused the inability to end call via monitor control., and once the call ended, the BT music operation was stuck paused and could only get it to play again via the phone control. ( no response from inputs from monitor or steering wheel )
apart from this, it seem to perform well. 
audio volume change between music and phone call is bang on with my setup. - big improvement!
I'll continue to use and report..

Thanks

dertimaushh, 2020-11-19:
Just did a quick test of 1.17 Beta - Caller ID is working. 

The Blinker "Numbers" are missing when I set the language to German. 

[image]

BR,
Tim

tedsalmon, 2020-11-21:
@dertimaushh,

I have updated the translations in the code -- thanks for the contribution! :)

@E46er

> I had 2x incidents where the screen changed to the music dashboard whilst on a phone call. - this caused the inability to end call via monitor control., and once the call ended, the BT music [...]
apart from this, it seem to perform well.

If you can reproduce this issue, can you get me a log? https://github.com/tedsalmon/BlueBus/wiki/Gathering-a-Log

Thanks!
-Ted

E39NLA, 2020-12-22:
Is there any chance of getting coolant temp display on E39's with the MID?

tedsalmon, 2020-12-22:
@E39NLA 

This is planned.

Thanks!
-Ted

tedsalmon, 2020-12-22:
I'm going to close this one @E46er -- If you reproduce the issue you mentioned, please open a new issue :).

Thanks again!
-Ted

closed 2020-12-22 by tedsalmon