tedsalmon/BlueBus #55: E39

socoolbreeze, 2020-11-21:
FW: 1.1.16
[image]
-Shows on positions 4,5,6: no disc status

-When using siri you hear audio switching (plops,tiks) only when using coax digital option

-DSP not active? When changing the echo or eq it's not changing.

tedsalmon, 2020-11-21:
@socoolbreeze 

Yeah, the BM24/BM54 offers menus that our North American BM53 does not (our radio overlay screens are empty). I will append more newlines to clear those menu options

For Siri, it is normal to hear a little bit of noise when switching over, but it shouldn't be bad. Can I get a record, if it is bad?

Thanks,
-Ted

socoolbreeze, 2020-11-22:
Hi Ted,

Thanks for clearing.
And when a title is too long it cannot continue on the right side?!

Switching wasn't noticeable when using SES. But then I didn't use SMB coax input.
For the ear it's noticeable but on video recording you don't hear it. Is a quick volume dip or something possible?!
(as a video editor I have to make a fade when most music ends....This sounds almost the same)

Thanks for the hard work so far!

tedsalmon, 2020-11-25:
@socoolbreeze 

Can I have you test this out? https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_17_beta_2.hex

Thanks!
-Ted

socoolbreeze, 2020-11-25:
[image]
That works 👍. Got a few tweaks for dutch language because of limited characters. Should I adjust in document?

tedsalmon, 2020-11-25:
@socoolbreeze 

Excellent! On language changes, please do update the document. If umlauts are used in the language, I can include them -- we are not limited to ASCII.

Thanks,
-Ted

socoolbreeze, 2020-11-26:
Ok I will make the changes asap because of character limitations.

One question about this.
[image]
After knipperlicht: you should see the numbers of blinks you can choose. Now it doenst show anything most times or a v, like the image.

Furthermore when booting the BC you see a flash of the before cleared situation. Don't know if that has to do with order of coding?! 
And when choosing tone and I come back to bluetooth screen you see the nodiscs does come back. After pressing display button it disappears.

socoolbreeze, 2020-12-02:
Had a few day's trouble with reaccuring screen[image]
This morning no troubles anymore. I'll monitoring it. Is a log helpfull?

EGascon, 2020-12-04:
> One question about this.
> [image]
> After knipperlicht: you should see the numbers of blinks you can choose. Now it doenst show anything most times or a v, like the image.
> 
> * In the translation document I saw that i noted :%s and it has to be %d so this is probably it!

I've got the same issue on my CD53: it displays "Blinkers:" but then does not scroll to show the number.

socoolbreeze, 2021-01-25:
Simetimes the overlay screen lines come back after using siri or on startup. Tried different apps, reset bluetooth etc. Still random.

Eventually it became very slow, when skipping to next number. The title, artist & album text freezes....
past 2 weeks and did a downgrade to stable FW release that works better now.

Don't know if my new OBDII port scanner from Nonda with Zus ios app interferes with the bluebus communication? Don't have a problem after downgrading yet...

On the previous comment about Knipperlicht AKA Blinkers the change of the :%s into the %d I would like to test it out. Iv'e shorten some text, but should be within max characters now.

greetz Ben from Netherlands

tedsalmon, 2021-01-26:
@socoolbreeze 

What do you mean by "overlay screen"? Are you talking about the default "P4. P5. P6." lines?

What firmware were you on before the downgrade? 

What are you on now?

Thanks!
-Ted

socoolbreeze, 2021-01-27:
Now using 1.1.16, it's the official stable version on your website.

The one I tested was the one linked in the thread (1_1_17_beta_2.hex) with Dutch translation and the solution for default lines p4,p5,p6 indeed.

The 1.1.16 works stable without clean lines and translation.

Strange that in beginning 2 months ago when started using beta 2 it worked fast and since few weeks slower. I cannot pinpoint why.

greetz

tedsalmon, 2021-01-28:
@socoolbreeze 

>Strange that in beginning 2 months ago when started using beta 2 it worked fast and since few weeks slower. I cannot pinpoint why.

I suspect that this has less to do with the BlueBus and more to do with the car as none of the code that drives the writing of the UI has changed.

Can you try the latest? https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_17_10.hex

Thanks!
-Ted

tedsalmon, 2022-06-29:
@socoolbreeze

Can you confirm if this is still an issue?

-Ted

socoolbreeze, 2022-06-29:
Hi @tedsalmon

It works okay. It's still visible sometimes the P overlays on the right side.
Usually after changing to other view or depending on song maybe.
Then after a song or two it disappears. And not all lines are filled. Sometimes only P4 and P6.
No consistency. I also want to try new fw with dutch translation (see other thread) but no new one to be found.

Thanks!
Ben