tedsalmon/BlueBus #56: BMW Z4 Coupe - unlock rear hatch

maupineda81, 2020-11-22:
As per the title, could it be possible to also unlock the rear hatch in the BMW z4 coupe along with the side doors when powering the car off?

tabor-a, 2020-11-23:
And the E46 Touring rear window and door.

tedsalmon, 2020-11-23:
Just so no one's hearts are broken, this all depends on the EDIBAS API and the original developers thinking ahead to make a diagnostic test for these locks 😁

tedsalmon, 2020-11-23:
Alright @maupineda81 and @tabor-a 

Since y'all want the feature, you're now in charge of helping to make sure that what I found is the correct command :smile:  --

`3F 00 00 0C 63 01 00`

This is listed as the `MERHK` job -> `Motor Entriegeln Heckklappe`. The door locks are currently actuated with the `MER` job, so  I suspect this is what we want.

Please follow this process: https://github.com/tedsalmon/BlueBus/wiki/Accessing-the-Terminal

Then, with the trunk locked and the ignition in position 1, run this command:

`send ibus 3F 00 00 0C 63 01 00`

Thanks!
-Ted

maupineda81, 2020-11-25:
I ted, I have run the command, all I get is an OK message

********************
~~~~~~~~~~~~~~~~~~~~~~~~~
BlueBus Firmware: 1.1.16
Try HELP or ?
~~~~~~~~~~~~~~~~~~~~~~~~~
# send ibus 3F 00 00 0C 63 01 00
OK
#
********************

but no physical change.

tedsalmon, 2020-11-25:
@maupineda81 

Thanks for testing -- I wonder if BMW considers it a "tailgate".

Please try:

```
send ibus 3F 00 00 0C 95 01 00
```

Thanks!
-Ted

maupineda81, 2020-11-25:
Same result. let me see if I can find a command in Tool32

maupineda81, 2020-11-26:
Hello Ted, I was able to find the central locking command in tools32... it is under ZKE5_S12.

I am attaching two files, one that has the command definition and the run of it, where you can see the HEX instruction (Central Locking Command.txt)

I am also attaching the GM5 module table, it is in german, but I will look into the jobs available.

Let me know if this works, or is useful. Without knowing how the current implementation works, calling the central locking switch may not be ideal, as if the car was open when switching the car off, it will close the doors, so I think the right way of doing it is by calling status, and only if close, the open.
[CentralLockingCommand-Z4.txt](https://github.com/tedsalmon/BlueBus/files/5600019/CentralLockingCommand-Z4.txt)
[GM5Table-Z4.txt](https://github.com/tedsalmon/BlueBus/files/5600020/GM5Table-Z4.txt)

tedsalmon, 2020-11-26:
@maupineda81 

Thanks for finding this! Try this:

```
3F 00 00 0C 03 01 00
``

In any case, I do track the central locking status. I will have a better look at all of this over the weekend.

Thanks!
-Ted

maupineda81, 2020-11-26:
it worked!!!

maupineda81, 2020-11-26:
After seeing the commands in GM5, would it be possible for the BB to fold the mirrors when turning the car off, and unfold at ignition in position 1? Just a thought.

BTW, I tried running the command from toolset32 but did not work, I may not be using the string correctly and ran out of time. I i usually do it in ISTA+ and then read the log file, that is how I found the unlocking commands.

tedsalmon, 2020-11-26:
Looks like unlocking tailgate is `3F 00 00 0C 02 01 00`, now that I look at the list of jobs again. When the BlueBus usually unlocks the doors, does the gas filler also unlock?

Mirror folding is covered in #24 ;).

Thanks!
-Ted

maupineda81, 2020-11-26:
> When the BlueBus usually unlocks the doors, does the gas filler also unlock?
> 
> Mirror folding is covered in #24 ;).
> 
> Thanks!
> -Ted

Hi Ted, no, it does not, just the doors, with the command you gave me yesterday evening (3F 00 00 0C 03 01 00), it did. It unlock the hatch and the gas filler cap.

tabor-a, 2020-11-26:
> When the BlueBus usually unlocks the doors, does the gas filler also unlock?
The fuel filler door unlocks when the driver door is unlocked from the key fob and remains unlocked until the driver door is locked again.

arrdem, 2023-05-19:
Verified

```
send ibus 3F 00 00 0C 63 01 00
```

as a working trunk unlock on my e85m