tedsalmon/BlueBus #57: New Setting Option: Keep Max Volume

vgergiev, 2020-12-07:
Starting from up [14.2 on iOS](https://piunikaweb.com/2020/11/23/ios-14-2-update-disables-option-to-turn-off-headphones-safety-setting/) iphone users can't disable headphone safety feature.
Whats impacted: when volume on Bluetooth device is set to max - iOS will reduce it to 50% from time to time to prevent ear damage. Users can't control it anyhow.

Suggestion: Is it possible for Bluebus to handle listen change event and set it back to max when the setting is enabled.

I'm always setting phone volume to max and controlling actual volume by car media controls.

Would be awesome if it's possible to get a volume change event without need to constantly raise the volume to max

EGascon, 2020-12-08:
No idea if that's possible, but I'd appreciate the feature too.

Whoever at Apple thought that was a good idea -_-

ShonkyCH, 2020-12-09:
My Galaxy S8+ does something similar and on top of that makes you acknowledge a warning to go above about 75%.

I need to set it up again but this should be fixable on Android using something like Tasker to detect the Bluetooth connection and then set maximum volume.

tedsalmon, 2020-12-22:
This should be possible, though I don't know if the iOS device will accept the commands to continue increasing the volume beyond their maximum preset.

Can someone try the `set vol` CLI command [on the attached firmware](https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_17_9.hex)? I am an Android user, and my only iOS device is a 6S not running the afflicted firmware :(.

You can test on a computer (without the car) easily with the following commands:
```
set ign on
```
<wait for BT to connect on your phone>
```
set vol
```

Repeat `set vol` until hopefully it goes beyond the limit. If this doesn't work, then there's nothing I can do :(

-Ted

vgergiev, 2020-12-23:
I've tested it on my iOS device and works perfectly!
I've just caught automated volume reduce and after that "set vol" changed volume from 50% to 100%.

I hope that fix will be possible, it's so frustrating to set the volume back each 25-30 minutes.

tabor-a, 2020-12-28:
Presumably this fix for iOS will also max out the volume for any Android device that has not already gone to max volume?

tedsalmon, 2020-12-30:
@vgergiev Thanks for testing!

@tabor-a - It's going to be a configuration option, but yes, it's device agnostic.

bpotter79, 2021-01-04:
I am wondering if this could be resolved by emulating a different CoD of the Bluetooth, making it appear as a car kit rather than just an audio device, 

this would also bring additional functionality, like apples “where did I park my car”

tedsalmon, 2021-01-05:
@bpotter79 

This is unfortunately not possible. I have no control over the Bluetooth stack as the BC127 doesn't expose any of those tid-bits.

:(

-Ted

dertimaushh, 2021-01-27:
[image]

Apple fixed this with iOS 14.4 - just tried this morning :-)

vgergiev, 2021-01-27:
Yes, I can confirm the same.
Note: if you've updated iOS to 14.4, but the configuration option doesn't appear for BlueBus, you need to connect to it and after that configuration will be added to device options.

I believe the issue can be closed, but maybe some of the Android users are interested in this option.

tabor-a, 2021-01-27:
> but maybe some of the Android users are interested in this option.

Yes, please keep it for us poor Android users!!!

tedsalmon, 2021-01-28:
Thanks for the update @dertimaushh!

I will keep this open as I want to add it as a default setting -- I get a lot of emails from users who don't have their phone volume maxed out and are thus not getting the best possible audio quality experience.

Thanks!
-Ted

tedsalmon, 2022-10-05:
This has actually been implemented since `1.1.18` I believe. It works well :)

closed 2022-10-05 by tedsalmon