tedsalmon/BlueBus #58: E38/E39/E46 Heated Seat Recall

E39NLA, 2020-12-07:
The heated seat switches (part of the "SZW" Switching Center Module) on 9/98 and later E38/E39's are connected to the I/K loop. I would think, therefore, that I-bus commands could be used to turn on/off the heated seats.

Could the BlueBus recall any heated seat settings as the car is turned off, then recall them when turned back on?

tedsalmon, 2020-12-07:
@E39NLA 

If you could provide some logs of you toggling heated seats, that would be great. Ideally you'd do this with the key in ignition position one to prevent loads of bus traffic.

I'll have a look through @piersholt's Tool32 dump too.

Thanks!
-Ted

E39NLA, 2020-12-07:
Sure! My E34 is asleep for the winter (although I intend to pull more logs on that as time allows) so I'll need some E39 development to keep me busy.

piersholt, 2020-12-08:
I don't believe you'll find much in the ZKE spreadsheet. @tedsalmon. Similarly, the logs won't yield much.

The switch centre _is_ connected to the I/K-Bus (if seat heating, active seats, or rear blind is equipped), but it's only for diagnostics. In the case of heated seats, the switch centre has dedicated IO to control the heating elements.

There's an SGBD for the switch centre. Presumably it would have status and activation jobs akin to other modules.

tedsalmon, 2020-12-22:
@E39NLA 

I've decided not to pursue this feature -- there's a lot of complication (constant polling to see if the heated seats are on before power is removed from the car) and I don't believe I can accurately implement it as I don't believe I can get/set the correct heating level.

In any case, thanks for the idea!

-Ted

closed 2020-12-22 by tedsalmon