tedsalmon/BlueBus #59: Behaviour during Traffic Announcements (TP) (European feature)

EGascon, 2020-12-11:
In Europe, radio broadcasters give regular traffic announcements. Our radios pick those up over RDS, and if the "TP" function is enabled, even in CD or AUX mode, the radio will pause playback and tune in to the radio station briefly and display the name of the radio station.

1. Currently the Bluebus overwrites the radio station name when in Bluetooth mode. For a truly OEM behaviour, this should not happen.
2. The Bluebus should pause the song during the announcement. I believe that this is not the case in the current 1.1.17.beta2 version.

Let me know if you need a log to identify the TP flag. Might be a bit tricky since one basically needs to wait until the radio station broadcasts a traffic report.

tedsalmon, 2020-12-12:
@EGascon 

Thanks for the report! Can you confirm that the BlueBus does NOT pause playback when TP announcements are made? I would expect the `RAD` to tell the `CDC` to stop or pause playback when making an announcement.

I will need a log in order to stop the BlueBus from overwriting the TP, unless of course this is a CD53 we're talking about, then it's unlikely there's any indication of a TP Announcement on the I/K-Bus.

Thanks!
-Ted

EGascon, 2020-12-12:
> @EGascon
>
> Thanks for the report! Can you confirm that the BlueBus does NOT pause playback when TP announcements are made? I would expect the `RAD` to tell the `CDC` to stop or pause playback when [...]

As I said, I believe it doesn't pause, but I'll make sure next time there's a TP. It's tricky to say since I'm usually driving when traffic announcements are made, so checking my phone to see if the music is played is not an option.

When I record that log for you I'll pay attention to music playback to see if it pauses or not. Unfortunately it is a CD53 so you might be right about the lack of info on the I/K-Bus :(

Thanks!

EGascon, 2020-12-14:
@tedsalmon

I can confirm that playback on my phone did not pause during the traffic announcement as I was driving along this morning. I'll try to gather a log next week.

Thanks!
-Eduard

tedsalmon, 2020-12-15:
> @tedsalmon
> 
> I can confirm that playback on my phone did not pause during the traffic announcement as I was driving along this morning. I'll try to gather a log next week.
> 
> Thanks!
> -Eduard

Thanks, @EGascon 

When you grab that log, can you make a note of the song you're listening to at the time so I have a frame of reference for events occuring around that time period? Otherwise, I have to look for unfamiliar bus traffic which is very tedious 🙂

EGascon, 2020-12-15:
> Thanks, @EGascon
>
> When you grab that log, can you make a note of the song you're listening to at the time so I have a frame of reference for events occuring around that time period? Otherwise, I [...]

Will do, no worries!

I'll also try to play a CD and see if playback is paused during a TA, for some reason I always thought it paused but now I'm beginning to doubt myself...

EGascon, 2021-01-05:
Discovered another implication of TP... I was on a call in Bluetooth mode when a Traffic Announcement kicked in. The call continued but I could not hear the other person until I cancelled the TA by pressing on the right option button on the CD53.

I know I still owe you a log, the holidays were busier than expected...

tedsalmon, 2021-01-06:
@EGascon

I bet this wouldn't be an issue if I broadcast the telephone status as "on a call", but then we're back to the issue of the volume not being manageable while on a call. I might be able to overcome this by firing off MFL -> RAD volume up/down messages for every volume up/down message I get from the RAD -> TEL.

I will test this hypothesis soon!
-Ted

EGascon, 2021-01-06:
> @EGascon
>
> I bet this wouldn't be an issue if I broadcast the telephone status as "on a call", but then we're back to the issue of the volume not being manageable while on a call. I might be able to [...]

I'm pretty sure it wouldn't be an issue if tel. status was "on a call" as you say. The volume not being manageable while on a call is an issue though.

As soon as I can gather a log of a TA I'll send it your way.

Thanks!
-Eduard

EGascon, 2022-06-18:
Hi Ted,

This comes way, way later than I would've liked, but hey, better late than never. I finally managed to pull a log during a Traffic Announcement. I just sent it to you by email.

Since I had to wait a while until they finally broadcast a TA, here's some data to help find the "right" line in the log:
- Song being played: I Love Rock 'N Roll by Joan Jett & The Blackhearts
- The TA started a little after after line 933,036 and finished before 1,003,256 in the log.

Fyi: I'm running firmware 1.1.19 RC12 in an '03 E46 Touring with a CD53 radio.

I hope I managed to pull the log correctly!

Many thanks!
-Eduard

piersholt, 2022-06-19:
@tedsalmon can you send me that log? 😃