tedsalmon/BlueBus #62: Static/scratching noise heard from speakers

bimmerluvr, 2021-01-07:
I recently received my Bluebus and connected it as per youtube videos. I have discovered a few issues.

First, there is a static noise present while the head unit is powered on and the source is set to "Bluetooth" and not playing audio as well as during playback. The noise is subtle while the engine is off and much louder with the engine running.

Today, I removed the rear panel again just to double-check all the connections. Everything looks good.  I noticed the ibus/power  cables were routed close to the audio/remote wire of an aftermarket amp I have installed. Thinking there may be some interference, I routed the cables away from the others. Tested again. No change.

Below are a couple of videos that show the issue while the key is in position I and II:

https://youtu.be/STgPINLZzVc

https://youtu.be/DT2ywxvu06I

Second, I opted for the TCU emulator cable. The BMW OE mic does not pick up any audio. The party on the receiving end only hears static. I really do not know how to troubleshoot here since it appears to be plug and play. 

Third, I tried to upgrade the firmware on my unit and the firmware tool will not connect. I tried several times with two different USB cables with no luck. 

Thank you in advance for your help!

calmelb, 2021-01-07:
With the second part, do you actually have a microphone installed in your roof (for all non convertibles)? Take a torch/ flashlight and shine it into the little grill and see if there’s something there. 

Otherwise if it’s there take the panel down and look to make sure it’s properly connected in case it’s come loose. 

With the third issue, you may have been unlucky with 2 cables that don’t have the data connections on it, try and find one that’s used for mobile phone syncing or similar to make sure it actually has the required data pins

tedsalmon, 2021-01-10:
@bimmerluvr

>First, there is a static noise present while the head unit is powered on and the source is set to "Bluetooth" and not playing audio as well as during playback. The noise is subtle while the engine is off and much louder with the engine running.

The noise from the video is inherent to the system. Every car using analog audio has this noise present, but it's not the BlueBus's fault -- it's the I/K-Bus bleeding impulse noise (when the line gets pulled low to transfer a byte)  into the line level audio input from the CD Changer. To verify, unplug the six pin audio connector from the BlueBus and you'll note that the noise gets way worse. I recommend you max the volume on your phone and you'll never need to crank the volume high enough to hear this. If this car has Nav, you might be able to fix it with some creative wire routing.

>Second, I opted for the TCU emulator cable. The BMW OE mic does not pick up any audio. The party on the receiving end only hears static. I really do not know how to troubleshoot here since it appears to be plug and play.

If @calmelb 's suggestion is not true and you do have a microphone, then there may a licensing issue with the BlueBus (very unusual but it's just a software issue).

In terms of not being able to update the BlueBus firmware, if you're using Windows then you may need to ensure the FTDI VCP Driver is installed and Windows is recognizing it as a COM port. It could also be that the BlueBus case is getting int he way, so you may need to crack it open slightly.

Thanks!
-Ted

bimmerluvr, 2021-01-11:
Thank you for your responses thus far.

@calmelb My car did not come equipped with a factory microphone. I purchased one on eBay as per the recommendation on the  Bluebus forum thread. Interestingly, my mic was delivered with the original pigtail still attached. The one I received only had 2 cables whereas the same connector in my car has 4 cables. If in fact there is a software issue, is there a way to overcome this?

@tedsalmon Based on your explanation, I now understand the noise is normal. I did briefly disconnect the six pin audio connector the other day and can confirm the noise got way worse. I always max the volume on the phone also. I do not have factory nav so that's that. End of discussion.

I will confirm the correct driver is installed on my laptop and open the box if necessary. I will then try to connect again. 

Bluebus is a wonderful product! I love being able to add features such as the multiple turn signal flashing, it is one reasons why I made this purchase. I thought I could also "code" automatic door unlocking as well but I do not see that setting. Is that a current or future setting? I am fairly confident I read or saw it on a video somewhere!

tabor-a, 2021-01-11:
> the same connector in my car has 4 cables

Sounds like that might not be the microphone connector?  The microphone connector should be a black 3-pin connector.  Other connectors in the area could include the white HomeLink wiring connector. Take a look at this video:
https://www.youtube.com/watch?v=63ZTz_p0veA

bimmerluvr, 2021-01-11:
I still have the pigtail from the mic I received. I will compare the connectors but they both seem to be identical.

bimmerluvr, 2021-01-12:
@tabor-a I was able to confirm there are 3 leads in the connector on my car, not 4 as previously stated. The pigtail that came with the mic does have 2 leads. The connectors are similar just with different color wiring. I did dig around for additional cables and found the Homelink cable but nothing else similar that actually plugs into the microphone. 

[image]

tedsalmon, 2021-01-12:
@bimmerluvr 

Can I get the part number from the microphone? What year is your E46? Are you using the 54-pin adapter or SES adapter (One is black and the other is blue)?

> I thought I could also "code" automatic door unlocking as well but I do not see that setting. I am fairly confident I read or saw it on a video somewhere!

Yes, this is just not available through the UI yet. Please see #46 for that :).

Thanks!
-Ted

piersholt, 2021-01-12:
@bimmerluvr @tedsalmon 

I'd guess this microphone is from an E38 or early E39 given there's no shield.

[image]

tedsalmon, 2021-01-13:
@bimmerluvr 

If @piersholt is correct, and he usually is, then the part number ends in 810 (IIRC) and it does not work with the BlueBus.

Thanks!
-Ted

bimmerluvr, 2021-01-13:
@piersholt 

Thank you for providing the diagram and additional information. 

@tedsalmon 

Sure enough, the PN for the mic I have does end in 810. Interesting because I am usually very diligent with my research and I recall reading that the mic ending in 810 was the one to get. 

My car is MY 2005. I believe the connector is black (I'd have to double-check). 

[image]

tedsalmon, 2021-01-13:
@bimmerluvr 

Weird, I've never seen that part come up in the forum conversations. Here are the part numbers known to work: 84318380338, 84318380319 and 84316938762.

Thanks!
-Ted

bimmerluvr, 2021-01-13:
@tedsalmon 

Thanks for your help in troubleshooting so far and confirming the part numbers. I'll report back once I get the correct one installed.

tedsalmon, 2021-01-13:
@bimmerluvr 

Great! Let me know when you're good to go so I can close the ticket :smile: 

Thanks!
-Ted

piersholt, 2021-01-13:
> Thank you for providing the diagram and additional information.

No problem! 😊

> I didn't know the 810 part even existed until pretty late in the game when I encountered a customer with it. That's why it was never in the forum posts.

To be fair @tedsalmon, the 810 is pretty darn obscure. I couldn't even find a vehicle with it in the parts catalogue before @bimmerluvr provided the PN! It was [US only, and limited to early E38, and E39](http://bmwfans.info/parts-catalog/84318352810).

> Here are the part numbers known to work: 84318380338, 84318380319 and 84316938762.

@tedsalmon I checked my touring and I have PN [84 31 8 374 075](http://bmwfans.info/parts-catalog/84318374075/) which is a three pin shielded greenback like it's replacement [84 31 8 380 319](http://bmwfans.info/parts-catalog/84318380319/). 

Might that be another option for @bimmerluvr? (I don't think 075 has come up before? Maybe I'm the guinea pig?)

bimmerluvr, 2021-01-17:
Gentlemen,

I settled on PN 84316938762. It is the most readily available on eBay and least expensive. I Will chime in once it arrives. Thanks again!

tedsalmon, 2021-01-17:
@piersholt 

If you are able to test PN 84 31 8 374 075, I would very much appreciate that! I haven't run into the part, but knowing if it's compatible is important :)

Sounds good, @bimmerluvr !

-Ted

bimmerluvr, 2021-01-24:
Update: I received the mic ending in 762 and connected it last night. It works!!!

Thank you to all who contributed to this thread and helping me reach a solution.

tedsalmon, 2021-01-24:
Great to hear, @bimmerluvr !

closed 2021-01-24 by tedsalmon