tedsalmon/BlueBus #63: Request: Welcome Message

QSSaber, 2021-02-10:
This holds miniscule priority but modLight 2 provides option for a welcome message to be printed on the instrument cluster. 
I think it would be a nice little gimmick if it is added :)

Some stuff I found.
https://github.com/kmotov/ibus-mudule 
https://modbmw.com/welcome-messages-list/

piersholt, 2021-02-11:
If you were considering contributing @QSSaber, this could be a good feature to start with. @tedsalmon has already written most of what you need, and I can write documentation for the applicable commands.

QSSaber, 2021-02-12:
Yes, I can definitely work on this. I will be getting my own bluebus this month.

vgergiev, 2021-04-13:
@piersholt , it would be really great if you could write a short doc on how to set up dev environment and how to build the project. I don't have any experience with C\C++ (I'm working with Salesforce :) ), but it's pretty clear to me what I need to change to apply desired changes.
Thank you in advance!

rsporsche, 2021-04-13:
@vgergiev 
I guess you need to install this MPLAB X: https://www.microchip.com/en-us/development-tools-tools-and-software/mplab-ecosystem-downloads-archive
Also the XC16 compiler: https://www.microchip.com/en-us/development-tools-tools-and-software/mplab-xc-compilers
Then I could open the firmware/application directory as a project in MPLAB X and build.

piersholt, 2021-04-15:
@vgergiev

@rsporsche has pretty much summed it up!

The development environment is a bit more verbose than you'll be used to, but don't let that put you off. We're dealing with a platform that's ultimately significantly simpler than anything like Salesforce. 😄

QSSaber, 2021-04-19:
Phew.. been quite busy and still am.. but definitely a weekend job to tackle on.
Didn't even have time to setup the ide, might use my existing netbeans setup and dig around some plugins.. 

But on the welcome message function, Ted's premade function IBusCommandIKEText would work. But apparently it talks to the business radio? src is TEL and destination being IKE. Interesting... I guess writing to CD53 works as if high OBC cars writing to instrument cluster. I.e. same function is called on the IKE. 

So something like this..
/* 
* text could be presets similar to modLight 2 module but can be something that could be set on CLI as well. I think.
/ text to hex 
IBusCommandIKEText(context->ibus, text);
//time out 
IBusCommandIKETextClear(context->ibus);

tedsalmon, 2021-04-20:
@QSSaber 

So, the CD53 listens for the TEL -> IKE write command simply so the developers wouldn't have to implement new frames to display data on the CD53. It's worth noting that a High OBC car will never have a CD53 with a display on it, only a MID or Navi, and that's how they can get away with it.

You're right that you can simply hijack these functions, but there is nuance to these frames. Have a look:
```
Describing the text line in the instrument cluster: ADDRESS 80 HEX

Message format:

<68 17 80> 23 62 30 <Option> <Text in ASCII Hex> <XOR>

Note: All characters that can be displayed in the text line must always be sent, otherwise the “old” characters are not overwritten. Thus the message length is always “17H”.

The message can be viewed with the following options:

35 00      normal display

37 01      Display text between two red arrows

37 03      Display text between two red, flashing arrows

37 04      Display gong and text between two red, flashing files 

37 05      Display gong and text 

37 08      Gong I (without displaying a message)

37 10      Gong II (without displaying a message)
```

Thanks!
-Ted

piersholt, 2021-04-20:
@QSSaber

Coincidentally, Ted and I were discussing this recently. I'm happily without a CD53, but Ted has tested the module extensively, so I can relay his findings.

This is a great example of BMW's ability to bend an existing solution to suit a new application. 

As far as the telephone `0xc8` is concerned, it _is_ talking to an high cluster (IKE, IKI) `0x80`. The CD53 is listening for any display messages sent to the cluster by the telephone, e.g. `c8 .. 80 23 [...]`.

The other piece of the puzzle is the MFL `0x50`.

The CD53 also listens for the **R/T** (Radio/Telephone) [button press](https://github.com/piersholt/wilhelm-docs/blob/master/mfl/3b.md#radiotelephone-rt-0b0100_0000) that the MFL sends to telephone. Specifically, it listens for when the MFL tells the telephone that it's in radio mode. This is so it knows when to turn off the display.

The CD53 also emulates the MFL. Specifically, it tells the telephone it's in radio mode. This happens when the radio controls are being used, and will prevent the telephone overwriting the radio display when it's being used.