tedsalmon/BlueBus #64: (Feature Request) M3 SMG display function on 5hp19

sven358, 2021-02-11:
I don't know if this can be done, but I retrofitted a M3 SMG cluster into mine 325i,
But once you do that you will lose the Manual/sport display on your normal cluster.
and the SMG cluster is turned on but it's just doing nothing. you can't see in wich gear the gearbox currently is.

So i was thinking, if we could get the actual gear from the bus and echo the SMG signal on the bus, would it display the gear?
I would love that to work. 

Sven

ShonkyCH, 2021-02-11:
I don't believe the ability to read the current gear or set the gear on the SMG dash is available from the Ibus which is what BlueBus uses. The Ibus is mainly used for less critical items, like navigation etc. SMG uses CANbus for most/all of its communications with DME.

sven358, 2021-02-11:
I don't know, maybe the actual display isn't that "critical" to be on the main canbus, If it's not possible I guess it stops here. but then again if you didn't ask you won't get the answer.

ShonkyCH, 2021-02-11:
It's not available.

sven358, 2021-02-11:
Okay, Then lets close this one. thanks anyway!

closed 2021-02-11 by sven358

tedsalmon, 2021-02-11:
Thanks @sven358 for your request and thanks @ShonkyCH for fielding it! :smile: 

To put the final nail in the coffin, the I/K-Bus on the E46 doesn't get any of the _cool_ (and rightful so!) data that is available on the CANbus. That being said, the selected gear is available on the I/K-Bus by means of [`0x13`](https://github.com/piersholt/wilhelm-docs/blob/master/ike/13.md#gear-0b1111_0000--8), which 100% doesn't work on manual cars, but may work on SMG cars. Still, I cannot take this information and write to the gear position indicator LCD as the IKE doesn't allow that kind of manipulation.

Thanks!
-Ted

ShonkyCH, 2021-02-11:
Interesting. I have an E46 M3 SMG, but haven't ever seen that message on the ibus.