tedsalmon/BlueBus #65: Problems with calling using OxygenOS // Oneplus Nord

sven358, 2021-02-11:
I don't know if it is a bug or a issue with the software version since i'm using the latest version? (Bluebus_1_1_6)
In the list of contacts there is nothing to be shown. the latest callers stay empty. So it looks like the bluebus doesn't have access to the contact information on the phone. 

The volume when calling is so low I cannot hear the other person with the engine running because the engine makes more noise. I cannot increase volume on the steering wheel or the knob in the 16:9 screen. The bluetooth volume is always on MAX during this issue. 

Both issues are permanent and not intermitted. 

Or could this be depending on my phone? I'm using the default app wich is just the google caller app.
https://play.google.com/store/apps/details?id=com.google.android.dialer&hl=nl&gl=US

Looking forward to hear from you,

Sven

tedsalmon, 2021-02-12:
@sven358,

>In the list of contacts there is nothing to be shown. the latest callers stay empty. So it looks like the bluebus doesn't have access to the contact information on the phone.

This is not implemented, but it's absolutely do-able. Searching your contact list on a non-touchscreen interfaces seemed tedious, so that's why it has not been prioritized. It will be implemented in time, though!

>The volume when calling is so low I cannot hear the other person with the engine running because the engine makes more noise
>Bluebus_1_1_6

If you are on `1.1.6`, then this issue is totally expected as it was a bug way back when. I suggest installing `1.1.16` from here: https://www.bluebus.dev/software

Thanks!
-Ted

sven358, 2021-02-12:
I'm sorry, my mistake. didn't know.

So I tried installing 1.1.16 today, but it was taking the UI way too soon back couldn't even change some settings.
Instead I installed 1.1.15, This one seems to work well, expect with calling there is this insane low quality calling with kinda a robot voice attached to this, I'm really curious if this even is the bluebus itself.

Sven

tedsalmon, 2021-02-17:
@sven358 

Try this:
https://github.com/tedsalmon/BlueBus/wiki/Accessing-the-Terminal

When you have the terminal open, run these commands:

```
bt cvc on
bt mgain c1
```

Thanks!
-Ted

sven358, 2021-02-23:
Thanks Ted, 

this worked for me. 

Sven.

closed 2021-02-23 by sven358