tedsalmon/BlueBus #66: CD53 "Bluetooth" disappears after pausing or skipping a song

josephfriedman, 2021-04-13:
When meta data is turned off, pausing or skipping/restarting a song leaves the CD changer text on display. "Bluetooth" only reappears after turning meta data back on and then off. Using version 1.1.17. Attached is a photo of the display. 
[https://share.icloud.com/photos/02pgk_wKnOse8_fT-hmb6v9qw](url)

rsporsche, 2021-04-13:
@josephfriedman FYI, you can attach images directly to the issue, no need to link to anything external.

tedsalmon, 2021-04-13:
@josephfriedman 

This is a complicated issue. Certainly I could continue to write Bluetooth back to the screen, but then I would also block you from viewing the time, changing the balance/fade/bass/treble and in some cases, the raido mode. The CD53 is not forthcoming with regards to it's screen state, so I have no idea that you're trying to perform actions on it, just that the screen has been overwritten and I need to write it back out. There may be room for improvement here, but the current functionality needs to be kept for certain use cases.

Thanks!
-Ted

tedsalmon, 2022-10-06:
After pondering this issue for more than a year, there really isn't a better way to implement this. Since I can't know what you're doing to the CD5x, my only choice is to depend on the end user to toggle the BlueBus UI manually :(

closed 2022-10-06 by tedsalmon