tedsalmon/BlueBus #67: Compatibility with the BMWRaspControl

BCKouma, 2021-04-19:
Heya,
since much longer before I got my BlueBus I have installed a modified TV-module with a rapsberry pi inside it, hosting the BMWRaspControl software and an interface.

SInce I got the BlueBus, I configured the Pi to act as an AUX-device, using the TV-module to forward the picture of the Pi to the 16:9 screen. Works so far... EXCEPT one little detail:

The Pi plays the music and uses the multifunction keys as a CD-changer would and with the BlueBus installed, the Pi gets a "pause" signal all the time. So I can start a song and it will play for just a second, then gets aborted by the BlueBus sending something like a pause- or stop-command - in theory.

Is there workaround for this?

With kind regards, 

Alex from germany

tedsalmon, 2021-04-20:
@BCKouma 

Are you sure the BlueBus is interfering? I don't emulate the Radio, which is the module that controls the CD Changer playback state. Can I get a log of this occurring? https://github.com/tedsalmon/BlueBus/wiki/Gathering-a-Log

-Ted

BCKouma, 2021-04-27:
Sorry for the long wait. Gonna try to log tomorrow. Work got the better of me these days. While the BlueBus is disconnected from the Ibus, the Rasperry Pi functions as normal. While plugged back in, the pi will immediately pause the playback and seems to get the pause-command every second or so.

tedsalmon, 2022-06-29:
> Sorry for the long wait. Gonna try to log tomorrow. Work got the better of me these days. While the BlueBus is disconnected from the Ibus, the Rasperry Pi functions as normal. While plugged back in, the pi will immediately pause the playback and seems to get the pause-command every second or so.

Just wanted to follow up about this before I close the ticket :)

-Ted

tedsalmon, 2022-10-06:
Closing ticket due to lack of response. Please let me know if you still want to troubleshoot this :)

closed 2022-10-06 by tedsalmon