tedsalmon/BlueBus #68: (Feature Request) reverse mirror tilt

MatthiasIwer, 2021-04-26:
Hi, 
is there a way to add a reverse mirror tilt as soon as reverse is engaged? Somthing like: on reverse tilt the miror for 2,3,4,5... seconds. And when reverse is out, then tilt the mirror back up the same seconds?
This would be a very nice comfy feature, thanks for feadback in advance, 
regards :)

rsporsche, 2021-04-26:
Isn't this basically a standard feature on BMWs? Well maybe not the 'for xyz seconds'. At least on my E39 the passenger side mirror will tilt when reversing if I leave the mirror adjustment selector switch in either the driver or passenger side (I forget which)

MatthiasIwer, 2021-04-26:
Hi, 
yes, its a standard feature, if you have electronic seats with memory. In case if not, then you have to tilt manualy ;) My z4 unfortnuate dosn´t have memory seats

rsporsche, 2021-04-26:
Ok, thanks I didn't realise that.

tedsalmon, 2021-04-26:
@MatthiasIwer

This is unfortunately not possible. Very few ZKE modules support mirror position alteration commands over the I/K-Bus, and even then I don't know the mirror position to begin with in order to know where it's going to end up when it tilts.

I think @piersholt may be able to keep me honest on this one, since he's a good bit more familiar with ZKE diagnostics.

Thanks!
-Ted

piersholt, 2021-04-27:
@MatthiasIwer 

I am (for better or worse), a good bit more familiar with ZKE as @tedsalmon mentions.

I can tell you, with respect to diagnostics, ZKE is the most convoluted system of the entire vehicle. 

ZKE3 (E38, E39 etc) for example, as it's very own, private bus (P-Bus)! ZKE5 (E46 etc), did away with the dedicated ZKE bus, but did little to simply the ZKE architecture. 

I'll give you an example. There's eight light module variants for our generation of vehicles. So when the lighting comfort features were developed, we had to content with up to eight different sets of diagnostic commands for a given feature. For ZKE there's..., well, I stopped counting at 20.

As for your suggestion @MatthiasIwer. There are diagnostic commands available to adjust the mirrors, however, putting aside the time required to reverse engineer the diagnostics, a tilt feature could not be implemented reliably. 

As @tedsalmon mentioned, in the absence of memory, there's no way to establish the position of the mirror, so we're flying blind.

"Okay, how about an arbitrary 2 seconds down, and 2 seconds up?". Good thinking, love where you're head is at, but this is also no good. There's a variable in play here- voltage. For a given input (i.e. 2 seconds), as the vehicle's voltage varies, the magnitude of the adjustment will change.

"Okay, it's possible to get the vehicle voltage, so we could then vary the input!" Again, solid reasoning, but the input is in seconds, thus we lack the precision required to allow for changes in voltage.

Long story short, the cost/benefit ratio is skewed in the wrong direction on this one.

(In the absence of volunteers to try and develop this, the implementation would be up to me, so I'm politely closing this one.)

closed 2021-04-27 by piersholt