tedsalmon/BlueBus #69: Pixel 4 bluetooth connectivity issues

stemo76, 2021-04-26:
Seems that my phone won't always connect on the first wake-up.  Red light in top corner of the radio.  Next time I start the car it connects.  

Not sure if this is just a Pixel issue or if others experience this as well

tabor-a, 2021-04-26:
What version of the firmware do you have? I had that problem with older versions but not with the latest versions.

tedsalmon, 2021-04-26:
@stemo76 

This sounds more like the "cold BlueBus" issue than anything. Is it less than 60F where you are when the issue first happens? Some BlueBus units fail to boot all the way when they're cold. I can fix them by removing three small components and replacing them with a mod-board that changes the power supply sequencing for the BC127 BT module.

The way to know for sure if you have the issues:

```
1. Remove the unit from the car and place it in the freezer for 60+ minutes.
2. Remove the unit from the freezer and immediately plug it into the USB port of a computer, make sure NOT to unplug it.
3, Follow this guide: https://github.com/tedsalmon/BlueBus/wiki/Accessing-the-Terminal
4. Run these commands:
set log bt on
bt version
5. Post the results
```

Thanks!
-Ted

stemo76, 2021-04-27:
> What version of the firmware do you have? I had that problem with older versions but not with the latest versions.

My unit came with the 1.1.17 installed.

stemo76, 2021-04-27:
> @stemo76
>
> This sounds more like the "cold BlueBus" issue than anything. Is it less than 60F where you are when the issue first happens? Some BlueBus units fail to boot all the way when they're cold. [...]

Thanks, I will give it a try tonight, I am in Western Washington, 60 degrees would be a nice day here.

stemo76, 2021-04-28:
> @stemo76
>
> This sounds more like the "cold BlueBus" issue than anything. Is it less than 60F where you are when the issue first happens? Some BlueBus units fail to boot all the way when they're cold. [...]

Cold out of the freezer I got this
# set log bt on
OK
# BT VERSION
[71458] DEBUG: BT: Send Command 'VERSION'
OK

warm out of my car I got this
# BT VERSION
[208932] DEBUG: BT: Send Command 'VERSION'
OK
# [208955] DEBUG: BT: Sierra Wireless Copyright 2018
[208959] DEBUG: BT: Melody Audio V7.3
[208964] DEBUG: BT: Build: 1544637564
[208969] DEBUG: BT: Boot Complete
[208973] DEBUG: BT: Send Command 'STATUS'
[208978] DEBUG: BT: Bluetooth addresses: 20FABB0ED80D 51FA8615E89A
[208982] DEBUG: BT: Profiles: A2DP AVRCP HFP BLE SPP PBAP MAP TWS
[208987] DEBUG: BT: Codecs: SBC AAC
[208992] DEBUG: BT: OK
[208997] DEBUG: BT: STATE CONNECTED[0] CONNECTABLE[ON] DISCOVERABLE[ON] BLE[IDLE]
[209001] DEBUG: BT: Got Status CONNECTABLE[ON] DISCOVERABLE[ON]
[209006] DEBUG: BT: OK

tedsalmon, 2021-04-28:
@stemo76 

Yeah, that's failing to start when it's cold. Can you please shoot me an email at support@bluebus.dev and I'll give you my address so you can mail it in for service.

Thanks!
-Ted

closed 2021-04-28 by tedsalmon