tedsalmon/BlueBus #70: Anyway to rename the bluetooth device to something else?

Rekpoint, 2021-05-02:
Anyway to rename the Bluetooth device to something else?
As the title states. Any way to change the Bluetooth name from "Bluebus" to something else?

ShonkyCH, 2021-05-02:
"BT NAME <name>" from the terminal ( https://github.com/tedsalmon/BlueBus/wiki/Accessing-the-Terminal ) will do it.

tedsalmon, 2021-05-12:
Thanks for answering @Rekpoint, @ShonkyCH :1st_place_medal: 

-Ted

closed 2021-05-12 by tedsalmon

shaylarihosain, 2022-08-31:
Hi! I was able to connect to the device through Terminal, but I don't actually know what command to run to change the name? 

I did try the HELP command, but each time I try it, the list of available commands is severely truncated (usually cutting off within the first or second line).

tedsalmon, 2022-08-31:
> Hi! I was able to connect to the device through Terminal, but I don't actually know what command to run to change the name?
> 
> I did try the HELP command, but each time I try it, the list of available commands is severely truncated (usually cutting off within the first or second line).

This is possible, but it's not straight-forward on the Hardware Version 2.x boards --

Here is the provisioning pack: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/BlueBus_BM83_Configuration_Tools.zip

You'll want to unzip it, then open Config_Tool/is208x_config_gui_tool v1.3.23.exe

From there, load BM83/BlueBus_Config_v4.hex

You'll want to click "Edit". On the new screen hit "Next". On the new screen click "MFB Power ON/OFF" (this setting always resets, so no matter what you need to set it back to MFB). Then click "Sys. Setup2" and device name will be the very first option.

Click "Finish" once done. The DSP config tool will now load. Click "Voice Function" then "AEC/AES" and ensure that "AES On/Off" is checked. This is another one that needs to be enabled but never "sticks" in this config tool.

After that, click Save. Then hit "Exit" and save the file once prompted.

Now, open [a terminal to the unit](https://github.com/tedsalmon/BlueBus/wiki/Accessing-the-Terminal) and issue the `BOOTLOADER` command. Now double click `bt_dfu.bat` in the root of the zip. After that, open `isupdate.exe`

From the "port" drop down, select the COM port that the bt_dfu.bat script told you it found the BlueBus on. Set "image num" to 1 and click "Connect". Once connected, click browse and find the configuration file, then hit "Update".

It will prompt you about clearing the paired devices list, but your answer is irrelevant as it will reset it no matter what you select (yay, another bug in their software!).

shaylarihosain, 2022-08-31:
Sorry, do I need a PC for this? I'm on a Mac so I'm not sure how to open .exe files.

tedsalmon, 2022-08-31:
> Sorry, do I need a PC for this? I'm on a Mac so I'm not sure how to open .exe files.

Microchip doesn't release macOS compatible binaries, so you will need a Windows machine. :(

shaylarihosain, 2022-08-31:
Got it. I think I might have access to a Mac with a Parallels installation of Windows 10. Thanks for your help!