tedsalmon/BlueBus #72: No sound when calling with bluetooth

sven358, 2021-05-12:
So i have no sound when calling i hear two clicks like the speaker is being powered but then nothing happens anymore. However if im using Google assistent It works fine. Any solutions?

I'm on 1.1.17

Sven

socoolbreeze, 2021-05-12:
Same version and same problem, but without clicks when using analog.
The caller hears the internal built-in microphone but the caller isn't audible on speakers.
Also not after switching bluetooth settings or re-pairing.

Ps. when switching handsfree off the callbutton on stear doesn't commands siri. Bug or feature as apple would say. Pause song works though.

Not-on-topic
I've added some NL/Dutch characters updates in file 🙏. Because of length and logic.

And got two menuerrors when language is set to NL

-when changing the amount blinkes..it could be set but no amount visable. And also a second setting in NL language. 

-When bleutooth pairing is sett off it's _on_ and vice versa.

greetz Ben from NL

tedsalmon, 2021-05-17:
@sven358

Can you make sure that Mic Bias is on and that Mic Gain is set to -23dB?

@socoolbreeze,

A lot to unpack here -- 

>Ps. when switching handsfree off the callbutton on stear doesn't commands siri. Bug or feature as apple would say. Pause song works though.

Can you explain this further? Does the Talking Head button ever summon Siri?

>I've added some NL/Dutch characters updates in file pray. Because of length and logic.

Thanks, I'll check for these updates and apply them.

>-when changing the amount blinkes..it could be set but no amount visable. And also a second setting in NL language.
This should be fixed in 1.1.17 final.

>-When bleutooth pairing is sett off it's on and vice versa.
So they're backwards? What's on/off in Dutch? Perhaps they are backwards. :D

Thanks!
-Ted

socoolbreeze, 2021-06-17:
Sorry for late reply. Got a day off and tested a lot. What worked for me was to connect to bluebus via terminal an did a full reset. Then installed FW 1.1.17 and Siri worked again as in the first FW release from the beginning.
Also it work better now with de blank lines on the right. When a disc number pops up it disappears after one song or so.

-Siri-
Command Siri using handsfree option **_on_** in menu works perfect.
When switching it **_off_** I cannot summon Siri.
In **_both_** cases I can pauze and play music.
This is probably because of HFP switching....simple audio streaming only?
[https://vimeo.com/564064661/53bf9fe4db](url)

-Dutch/NL updated translation-
Thanks for applying them. If you could give me an FW update.

-Blinkers amount (in NL/Dutch translation)-
I am using the final 1.1.17 and that still gives a * or something.

-Bleutooth pairing setting-
In Dutch
On: Aan
Off: Uit
and it's backwards yup. (the translation document confirms translation)

greetz Ben

adethefade, 2021-07-20:
My initial phone-related problem (it dialled but no audio either way) was largely resolved by running the CVC tool as suggested by Ted. Now I can make calls and hear the person I've called, but that person can't hear me. The original in-car phone system uses the the installed mic and works fine. 
Any suggestions welcome.

Many thanks, Ade.

tedsalmon, 2021-07-29:
@adethefade 

We figured this out, right? Your car was wired for Siemens rather than the traditional 54-pin connector, right?

-Ted

adethefade, 2021-07-29:
Hi Ted. Yes. The revised pin connections you kindly advised me on dealt with this problem. As you say, it was unique to the Siemens TCU.

Many thanks

Adrian

closed 2021-08-07 by tedsalmon