tedsalmon/BlueBus #73: Turn off the bluetooth power saving mode

KimParking, 2021-05-29:
Hi. I'm really happy with Bluebus.
But I have a problem with bluetooth.

When I drive, I usually turn on the GPS(navigation) app and It tells me a direction, warning sound for speed radar.
If when I play a music, the sound of GPS app is played normally.
but If when I don't play a music, the sound of GPS app in the front part is skipped.
eg. At the next intersection, turn right. → (~skipped~) intersection, turn right.

I think this caused by power-saving mode of the bluebus's bluetooth module.
To avoid this problem, I play the mp3 files with no sound to prevent bluebus goes in power-saving mode

tedsalmon, 2021-05-29:
Hi @KimParking!

Yes, the Bluetooth audio stream gets suspended when you're not playing anything after ten seconds and when it starts again, there's a 1-2 second lag. The unfortunate part is that I have no control over this as it's done by the BC127 directly. I've thought about ways to fix this but none of them are viable, such as playing zero volume tones while nothing is playing. This results in pops in the speakers :/.

I'll keep this open to keep me on the trail for a fix.

-Ted

tedsalmon, 2026-02-24:
This was ultimately fixed by changing hardware. A V2 unit does not have these issues.

-Ted