tedsalmon/BlueBus #74: CD53 Controls not working right and crashes

Tom192, 2021-07-16:
Hi Ted, unfortunately when I try to press anything the display goes to the CD 7-1 for example either for half a second or it stays there and bluebus crashes eventually. The main problem is that I can't control anything unless I press every button 2 times quickly but sometimes it also crashes while doing this.

I couldn't find anything similar in the forums. I hope you have an idea. I created a log where i tried to press forward (or next song) on the radio. I once forwared a song on my phone.

If you need any additional info just let me know !

Regards 

Tom

tedsalmon, 2021-07-16:
@Tom192,

Sounds like you have a CD43 -- does it have a "CD" button rather than "MODE"? These were standard on very early E46s. In that case, you have to double press each button to get an action to fire off -- the first will dismiss the text on screen and the second will perform the action you want.

Can you please explain what you mean by "crash"?

Thanks!
-Ted

Tom192, 2021-07-16:
@tedsalmon 

Thanks for your quick answer! Yes you are right. It looks like a cd53 from the outside but it has a CD Button like you mentioned above. Ok so knowing that helps a lot ! :)

Sometimes when I tried to change something in the settings it seemed to have crashed because the radio changed to radio by itself and I couldn't get back into Bluetooth. I had to unplug the 3 Pin and Plug it back in to "restart" it.
This happend 2 or 3 times but right now it is working normal. I guess everything is fine now.

Thank you very much !

closed 2021-07-16 by Tom192