tedsalmon/BlueBus #76: Question: How to setup dev environment and project for bluebus?

vgergiev, 2021-07-29:
Hi everyone,
I'm new to this type of development and curious on what is the proper way to setup an IDE/project to build a new version of the firmware.

I know that I need to install:
1. MPLAB X: https://www.microchip.com/en-us/development-tools-tools-and-software/mplab-ecosystem-downloads-archive
2. XC16 compiler: https://www.microchip.com/en-us/development-tools-tools-and-software/mplab-xc-compilers

But after installation on the new project screen, it asks me for family and device.
Could someone please send me some guidance on how to open this repo in MPLAB IDE and build a new version of firmware?

Sorry for asking such stupid questions, but don't know anything about embedded development.

Thank you in advance,
Vlad

piersholt, 2021-07-30:
@tedsalmon, I'm happy to look at this.

@vgergiev are you Windows or macOS?

vgergiev, 2021-07-30:
Thank you!
I'm using Windows.

dertimaushh, 2021-07-30:
Following this one closely - as I would also like to get into this topic :-)

piersholt, 2021-07-31:
Okay, so, the [BlueBus repository](https://github.com/tedsalmon/BlueBus) contains several interrelated projects.

- `./enclosure` is the BlueBus 3D printed case
- `./firmware` is the BlueBus software
- `./hardware` is the BlueBus PCB
- `./utility` is the BlueBus firmware utility

You might be thinking this is self explanatory. I point it out as you might be used to cloning a git repository, pointing an IDE at it, and _voilà_, but that's not the case here.

It's the firmware we're interested in, so let's take a closer look at `./firmware`!

You'll find the following directories:

- `./firmware/application`
- `./firmware/bootloader`

Okay, so fun fact, there's actually **two** distinct pieces of software running on BlueBus. This is what we're seeing in this directory.

1. `bootloader` is a bit like BIOS
2.  `application` is a bit like Windows

We only need to worry about `application`. That's the heart and soul of BlueBus- not discounting Ted, of course.

So, in summary, we care about `./firmware/application/`

piersholt, 2021-07-31:
Now, working under the assumption you've installed both MPLAB X and XC16... 

- open the IDE
- select _File > Open Project..._
- navigate to your local copy of the BlueBus repository
- drill down to `./firmware/application`
- select _Open_

You _should_ see something like this appear:

[image]

piersholt, 2021-07-31:
I'd recommend ensuring you can get this far before I go into further detail. It might only be a couple of steps but with different platforms, software versions etc, it's worth checking there's no problems at this point.

vgergiev, 2021-08-02:
@piersholt , thank you I was able to open the project with the same structure (don't know why but the "Open project" window was incredibly slow).

P.S.
Sorry for the long reply, I was skydiving over the weekend :)

piersholt, 2021-08-04:
Well, after skydiving, this shouldn't be too scary! 😄 

Give me a day or two, then I'll go into more detail.

vgergiev, 2021-08-18:
Hi @piersholt,
Could you please give us the next part of the instructions? :)

tedsalmon, 2024-01-28:
Closing as this is complete.

closed 2024-01-28 by tedsalmon