tedsalmon/BlueBus #77: Call Volume -

dertimaushh, 2021-07-30:
Hi Ted,

I finally nailed down the issue with my call volume. Running the latest FW 1.1.17.

Calls came in very quiet - even when previous Bluetooth Audio was very loud and the phones volume was maxed out. 

In order to raise the call volume I had to turn the volume knob at least 20 time before anything changed. It took 100 turns in order get the desired call volume which makes adhoc adjustments in a call close the impossible. 

Again for Audio all is fine. 

Any ideas?! 

Best regards,
Tim
 
Using an iPhone 11pro Max on iOS 14.7.1

blackfoton, 2021-07-31:
I have the same issue. Audio is good, but when a call comes in, I have to crank up the volume to hear anything. In my 2001 E46 330i Touring the call sound comes from the speaker below the steering column.

tedsalmon, 2021-08-07:
@dertimaushh

You have an E38, right? Does it have the DSP Amplifier or not?

When this issue crops up, are you in Bluetooth mode or AM/FM/CD/TAPE?

@blackfoton 

I've never seen a car with the speaker on the bottom of the steering column but I know they're out there (I bought one such speaker to test during development). I'm guessing this is a Non-Nav car as well?

-Ted

dertimaushh, 2021-08-07:
Hi Ted, 

no DSP, in Bluetooth mode. 

BR,
Tim

dertimaushh, 2021-08-10:
Any ideas?

blackfoton, 2021-08-10:
Hey Ted, yes Non-Nav car. I'll test it a bit further, if this only affects classic phone calls, or VIOP also (Facebook Messenger etc) and report back.

tedsalmon, 2021-08-12:
@dertimaushh,

Any change I can get a log of you making a call and changing the volume? There's no reason that audio should only increment slowly in Bluetooth mode. https://github.com/tedsalmon/BlueBus/wiki/Gathering-a-Log

@blackfoton 

Weird. You should NOT be getting audio from that speaker when you take calls in Bluetooth mode -- they should play over all speakers. Outside of Bluetooth mode, the system will likely use the single speaker.

Thanks!
-Teed

dertimaushh, 2021-08-12:
Will give it a try. 

It doesn't matter what mode i am initially in when I make or get a call. 

The volume behaviour is always the same - as described above. 

Btw is there a way to route calls via all speakers? 

BR,
Tim

tedsalmon, 2021-08-14:
@dertimaushh,

It should always route through all speakers if you're in Bluetooth mode, that's why I am a bit confused and want a log :)

-Ted

dertimaushh, 2021-08-16:
Issue Update after a quite evening in my car. 

Calls get routed via all Speakers in Bluetooth Mode and via a single Speaker outside of Bluetooth mode. Never realized before.

In calls outside Bluetooth mode all is good - I can increase and decrease the call volume ad hoc and all works fine.

The issue is only present in Bluetooth mode - which I am using like 99% of time. 

Still need to capture a log for that. 

Best regards,
Tim

tedsalmon, 2021-08-18:
@dertimaushh,

Did you happen to retrofit a Siemens CID radio from the E8x into your car? Those radios don't respond to volume changes while in TCU mode --  a bug I'm working on separately :)

Thanks!
-Ted

dertimaushh, 2021-08-18:
No, using a normal BM54 with flat pins.

blackfoton, 2021-08-19:
> @dertimaushh
>
> You have an E38, right? Does it have the DSP Amplifier or not?
>
> When this issue crops up, are you in Bluetooth mode or AM/FM/CD/TAPE?
>
> @blackfoton
>
> I've never seen a car with the [...]

Hi Ted!

Yes, it's a non-nav car.

Tested it today:
- calls are always routed to all speakers (I was wrong before: the speaker below the steering column doesn't do anything now)
- no matter how the call is made –incoming or outgoing,  phone call or VOIP( tested Facebook Messenger)–, the call volume (set to max on phone) is always way lower than the music volume
- my voice volume was fine on the other side

dertimaushh, 2021-08-21:
Hi Ted, 

Wanted to log my problem - and for some reason decided to re-flash the latest firmware. 

And now everything is working fine - lol. 😅

Will drive my car down to the Cote D'Azur next month - so plenty of testing time. 

Best regards,
Tim

tedsalmon, 2021-09-03:
@dertimaushh Hey Tim,

Great to hear! Please let me know if the issue reoccurs.

Thanks!
-Ted

closed 2021-09-03 by tedsalmon