tedsalmon/BlueBus #79: No sound on both ends while making BT phonecall

Marb06, 2021-08-07:
Two E38s build date 1999+

One E38 with the Motorola phone system where the antenna amplifier is next to the TCU. TCU housing is plastic. Partnumber on TCU module: 84216916565
The other E38 with the (Siemens?) black ''BMW'' wired phone with UTP cable and connector. Black metal TCU module with coax connection on the side. Partnumber on TCU module: 84116911042

Connected the DB25 connector to the car and Bluebus. 

Everything else seems to work fine. Calls can be answered or stopped with buttons on steeringwheel or buttons on the 16:9 or 4:3 screen.  Number/name display works. 
I can hear some very light crackling/change over in the front speakers when the call is made and when the call is answered and call is ended. There is no sound at all from both ends. Also no ringing/dialing sound.

Smarthphones: Iphone 11 and Huawei Mate 20 Pro

BT music streaming no problems at all. Works like a charm. 

Other question:

Telephone function via main menu does not work? (The telephone menu where you can also make an emergency call) I can't enter telephone numbers or go in the phonebook. I gave the Bluebus permission to use the contacts and contactlist from my smartphone

tedsalmon, 2021-08-10:
Howdy!

Some units did not receive the CVC license, which results in zero call audio on both ends. 

Fortunately I have a tool for this: https://github.com/tedsalmon/BlueBus/wiki/Clear-Voice-Capture

If the tool says "Working" for more than thirty seconds, close it out and try it in the car again.

Yes, the telephone menu is largely unimplemented. Displaying your contacts on the screen is not necessarily as easy as it sounds, unfortunately.

Thanks!
-Ted

Dashdot66, 2021-08-10:
Just tried on one of the two E38 mentioned above and it now works. Noticed however that when receiving a call no ring signal indicating the incoming call is heard. The screen does show the incoming call without the phone number, once the call is accepted the name (from my address book) of the caller is displayed.

tedsalmon, 2021-08-12:
@Dashdot66

The default configuration is to play the ringtone in-band, so that's weird. Can you follow this process: https://github.com/tedsalmon/BlueBus/wiki/Accessing-the-Terminal

Then run this command: `restore`

You'll need to pair your phone and BlueBus again after that.

Thanks!
-Ted

Dashdot66, 2021-08-13:
That solved it. Not sure what combination of settings caused that.

Thanks,
André

closed 2021-09-03 by tedsalmon