tedsalmon/BlueBus #81: Caller hears echo of their own voice

christrunk, 2021-08-24:
Hi there,

Here's the spec of my car:

> Car: E39 M5, 2003 (Dec 2002 build date)
> https://www.mdecoder.com/decode/cb37557

I'm using the factory fitted mic and speakers, but callers complain of hearing their own voice back as an echo on calls.

Is there anything I can do?

Many thanks,
Chris

tedsalmon, 2021-08-24:
Chris,

Please refer to this as the first step of troubleshooting: https://github.com/tedsalmon/BlueBus/wiki/Clear-Voice-Capture

Thanks!
-Ted

christrunk, 2021-08-31:
Hi Ted,

I knew you'd have the answer! :)

I've updated the CVC and got the message "Successfully Applied License", but callers say it's only "a bit better".

It's gone from "OMG I can't even speak to you it's so annoying" to "That's just distracting".

Is there something else we can do?

Many thanks,
Chris

bcagnoni, 2021-08-31:
Just thinking loud here: would mic gain setting have an impact on this? Default should be -23db according to the wiki page.

Sent from ProtonMail for iOS

dertimaushh, 2021-08-31:
What's the part number of your microphone? I think there one (newer version) that doesn't work with Bluebus algorithm that cancels the echo. Changed mine last year - and no more caller complaints afterwards :-)

dertimaushh, 2021-08-31:
Found this: 

I highly recommend the OE microphones with these part numbers: 84318380338 and 84318380319.

AVOID: 84316938762. It implements a much higher gain pre-amplifier which sometimes interferes with the BlueBus's echo cancellation algorithm and allows the caller to hear themselves over your speakers.

I'm not too sure what the difference between 338 and 319 are, honestly 

Thanks!
-Ted

dertimaushh, 2021-09-02:
Yes that looks all good.

tedsalmon, 2021-09-02:
@christrunk,

How loud is the in car volume on calls? At some point, the AEC is unable to distinguish echo from other parties in the vehicle depending on volume :/. Everything else looks great.

Thanks!
-Ted

christrunk, 2021-09-02:
> How loud is the in car volume on calls? At some point, the AEC is unable to distinguish echo from other parties in the vehicle depending on volume :/. Everything else looks great.

The volume is normal I would say, not particularly loud but naturally loud enough to go over road noise at 70mph

tedsalmon, 2021-09-03:
@christrunk 
> The volume is normal I would say, not particularly loud but naturally loud enough to go over road noise at 70mph

Hmm. Can you play around with the volume and see if that has any impact? What's the serial number of your unit? If you don't have Navigation to get that info, just tell me when your BlueBus was mailed out.

Thanks!
-Ted

bmos, 2021-09-07:
I just got my dad's 540i set up for bluetooth calls and noticed this same issue (although he has the not-recommended 84316938762). Serial was in the 500s, 545 maybe? His voice is also pretty quiet. Should I get him the older 84318380338 mic?

tedsalmon, 2021-09-09:
> I just got my dad's 540i set up for bluetooth calls and noticed this same issue (although he has the not-recommended 84316938762). Bluebus serial was in the 500s and it shipped sometime last summer (August I think). His voice is also pretty quiet. Should I get him the older 84318380338 mic?

I would go with 84318380319 off eBay. I've seen many have success with this microphone lately.

-Ted

shintopig, 2022-02-08:
I've emailed @tedsalmon about this but I figured I'd add it here as well.  Done the troubleshooting with no luck.

I bought 84318380338 and even with the most recent updates have that echoing problem.  So maybe avoid this one?  Unsure.

I'm going to purchase 84318380319 and update to see if that fixes things.

And I guess I'll avoid 84316938762.

(Just FYI I have a 99 E46 sedan)

christrunk, 2022-02-08:
@shintopig 

I never managed to resolve this one. I ran out of time and ended up buying some Airpods anyway!

I have all the updates and correct kit in the car, but lowering the volume did help.

tedsalmon, 2022-06-29:
To close the loop on this, there are a few things to keep in mind:

* Clear Voice Capture must be enabled via the tool
* -23dB Mic gain must be set
* The 338 and 319 microphones do tend to work better for most folks than the 762, though I have never had issues with any microphones personally.

If none of these things do it, then the only room for improvement lies in turning the volume down. Turn it down as much as you can while making sure you can still hear the caller. These steps haven't failed me yet, but it can be annoying to toy with the volume :(

-Ted

closed 2022-06-29 by tedsalmon

Keff789, 2023-06-25:
I have the same issue, is clear voice capture still needed/recommanded for HW2.0?

tedsalmon, 2023-06-26:
> I have the same issue, is clear voice capture still needed/recommanded for HW2.0?

No. What microphone are you using (Part Number) and what Mic gain are you at?

-Ted

Keff789, 2023-06-28:
Mic Gain is at the moment at 0 db and the microphone is a 8431-6938762. I have also a 8380319 lying around, that i could test. Regarding to your üpst from june 29 2022 would you recommend to switch to the 319 and set the mic gain to 23db? On my car it does not show -xx db