tedsalmon/BlueBus #85: Pressing radio power button during call breaks media playback after call

bmos, 2021-09-07:
Radio looks like this: https://a0.amlimg.com/ZjAzY2U4NDJkYTM5MzI5ZTM1M2JjODMwMDc2YzU2YjjRO8BsmF0Q7Pqof-d53YTSaHR0cDovL21lZGlhLmFkc2ltZy5jb20vZmQ0MjI4OWFjYmZjM2Q2NTBlNWYzZTRkYzliMWY0YmM1ZDc2MDRiMjQ3MTI2OTM4YjE1OWIzYjBlOTRjM2QzMy5qcGd8fHx8fHw3MDB4NTI1fGh0dHA6Ly93d3cuYWR2ZXJ0cy5pZS9zdGF0aWMvaS93YXRlcm1hcmsucG5nfHx8.jpg

Steps to reproduce:
1. Make phone call and press radio power button in-call
2. After call, try playing media. Media will play for half second or so and then pause.

To fix, must make call and then turn radio back on.

tedsalmon, 2021-09-09:
@bmos 

Huh. I never thought about hitting the power button mid-call. Is there a possibility I could have you send me a log of this? https://github.com/tedsalmon/BlueBus/wiki/Gathering-a-Log

I do not have access to a vehicle with the MID-based Business radio, and while I do have a bench setup, I'm unable to get any audio out of it.

If that's not possible, I'll go as far as I can on the bench :)

-Ted

bmos, 2021-09-09:
Can do. I should have a chance over the weekend.

bmos, 2021-09-14:
I was unable to reproduce it, but I did notice behavior that felt similar and weird.
This log is me turning on the radio, having media auto-play, and then making a phone call. during the call, i shut off the radio which left the call connected but made me unable to communicate. ending the call while in this state made the radio go dark for a couple of seconds and then it seemed to reboot or something similar. I  was then able to play/pause normally.
```
<html><body>
<!--StartFragment--><div class="gb_uf gb_of">

  |  
-- | --
 

</div><div class="a-b-Sh-ng a-b-va-Zf" role="main" style=""><div class="a-b-ah" style=""><div class="a-b-r-j"><div class="a-b-r-lc a-b-lc"><div class="a-b-Yk-hj" style="left: 1714.4px; top: 56px; text-align: left;" dir="ltr"><div class="Yk Yk-hj-S Yk-hj-S-yl Yk-tb-pf Yk-tb-Xi" id="Yk-hj-S" role="list"></div><div class="Yk-ra-no"></div></div><div class="a-b-r"><div class="a-b-r-Tl a-b-va-Zf" style="height: 9731px;"><div class="a-b-r-Ec a-b-tb-Ce" style="left: 205.6px; top: 56px; padding-bottom: 56px;"><div class="a-b-r-x" role="document" aria-label="Displaying bluebus.txt" tabindex="0" style="width: 1469px; height: auto; text-align: left;" dir="ltr"><pre class="a-b-r-La">=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2021.09.14 13:34:57 =~=~=~=~=~=~=~=~=~=~=~=
~~~~~~~~~~~~~~~~~~~~~~~~~
BlueBus Firmware: 1.1.17
Try HELP or ?
~~~~~~~~~~~~~~~~~~~~~~~~~
# [1104742] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 08 22 4E 
[1105146] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 00 22 46 
[1105405] DEBUG: BT: AVRCP_STOP 11
[1105542] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 08 22 4E 
[1105608] DEBUG: BT: AVRCP_PAUSE 11
[1105613] DEBUG: BT: Paused
[1105655] DEBUG: IBus: RX[23]: C8 15 C0 23 40 20 42 6C 75 65 74 6F 6F 74 68 20 50 61 75 73 65 64 1E [SELF]
[1105667] DEBUG: IBus: RX[11]: C8 09 C0 21 C3 00 40 7C 7C 20 83 [SELF]
[1105694] DEBUG: BT: AVRCP_PLAY 11
[1105699] DEBUG: BT: Playing
[1105704] DEBUG: BT: Send Command 'AVRCP_META_DATA 11'
[1105722] DEBUG: IBus: RX[10]: C8 08 C0 21 C3 00 40 20 3E BC [SELF]
[1105729] DEBUG: BT: OK
[1105819] DEBUG: BT: AVRCP_MEDIA 11 TITLE: Summertime (1962 Recording Remastered)
[1105927] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 00 22 46 
[1106127] DEBUG: IBus: RX[7]: 80 05 BF 18 00 00 22 
[1106193] DEBUG: IBus: RX[5]: C0 03 C8 01 0A 
[1106206] DEBUG: IBus: RX[6]: C8 04 C0 02 38 36 [SELF]
[1106317] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 08 22 4E 
[1106326] DEBUG: BT: Send Command 'AVRCP_META_DATA 11'
[1106333] DEBUG: IBus: RX[5]: 68 03 18 01 72 
[1106345] DEBUG: BT: OK
[1106349] DEBUG: IBus: RX[6]: 18 04 68 02 00 76 [SELF]
[1106390] DEBUG: BT: AVRCP_MEDIA 11 TITLE: Summertime (1962 Recording Remastered)
[1106395] DEBUG: BT: AVRCP_MEDIA 11 ARTIST: Pete Seeger - Topic
[1106400] DEBUG: BT: AVRCP_MEDIA 11 NUMBER: 0
[1106405] DEBUG: BT: title=Summertime (1962 Recording Remastered),artist=Pete Seeger - Topic,album=
[1106440] DEBUG: BT: AVRCP_MEDIA 11 TOTAL_NUMBER: 0
[1106446] DEBUG: BT: AVRCP_MEDIA 11 PLAYING_TIME(MS): 217000
[1106452] DEBUG: IBus: RX[30]: C8 1C C0 23 40 20 42 6C 75 65 74 6F 6F 74 68 20 53 75 6D 6D 65 72 74 69 6D 65 20 28 31 3C [SELF]
[1106686] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 00 22 46 
[1107087] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 08 22 4E 
[1107146] DEBUG: IBus: RX[9]: D0 07 BF 5B 00 00 00 00 33 
[1107152] DEBUG: OFF &gt; Unrelated activity!
[1107268] DEBUG: IBus: RX[14]: 80 0C E7 24 01 00 20 31 3A 33 36 50 4D 7D 
[1107475] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 00 22 46 
[1107885] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 08 22 4E 
[1108085] DEBUG: IBus: RX[7]: 80 05 BF 18 00 00 22 
[1108303] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 00 22 46 
[1108444] DEBUG: IBus: RX[30]: C8 1C C0 23 40 20 6C 75 65 74 6F 6F 74 68 20 53 75 6D 6D 65 72 74 69 6D 65 20 28 31 39 47 [SELF]
[1108688] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 08 22 4E 
[1108944] DEBUG: IBus: RX[30]: C8 1C C0 23 40 20 75 65 74 6F 6F 74 68 20 53 75 6D 6D 65 72 74 69 6D 65 20 28 31 39 36 1D [SELF]
[1109083] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 00 22 46 
[1109444] DEBUG: IBus: RX[30]: C8 1C C0 23 40 20 65 74 6F 6F 74 68 20 53 75 6D 6D 65 72 74 69 6D 65 20 28 31 39 36 32 5A [SELF]
[1109477] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 08 22 4E 
[1109880] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 00 22 46 
[1109944] DEBUG: IBus: RX[30]: C8 1C C0 23 40 20 74 6F 6F 74 68 20 53 75 6D 6D 65 72 74 69 6D 65 20 28 31 39 36 32 20 1F [SELF]
[1110080] DEBUG: IBus: RX[7]: 80 05 BF 18 00 00 22 
[1110140] DEBUG: IBus: RX[9]: D0 07 BF 5B 00 00 00 00 33 
[1110146] DEBUG: OFF &gt; Unrelated activity!
[1110274] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 08 22 4E 
[1110444] DEBUG: IBus: RX[30]: C8 1C C0 23 40 20 6F 6F 74 68 20 53 75 6D 6D 65 72 74 69 6D 65 20 28 31 39 36 32 20 52 39 [SELF]
[1110672] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 00 22 46 
[1110944] DEBUG: IBus: RX[30]: C8 1C C0 23 40 20 6F 74 68 20 53 75 6D 6D 65 72 74 69 6D 65 20 28 31 39 36 32 20 52 65 33 [SELF]
[1111079] DEBUG: IBus: RX[12]: 80 0A BF 13 02 00 40 00 00 08 22 4E 
[... 537 lines cut ...]
[1175736] DEBUG: IBus: RX[7]: 68 05 18 38 01 00 4C 
[1175744] DEBUG: IBus: RX[6]: 00 04 BF 72 02 CB 
[1175751] DEBUG: IBus: RX[6]: 80 04 30 1B 00 AF 
[1175761] DEBUG: IBus: RX[7]: 00 05 BF 7A 51 20 B1 
[1175768] INFO: Handler: Central Locks unlocked
[1175773] DEBUG: BT: AVRCP_STOP 11
[1175779] DEBUG: BT: CLOSE_OK 11 AVRCP C08C715EE985
[... cut ...]
[1175831] DEBUG: IBus: RX[27]: 30 19 80 1A 37 08 4B 45 59 20 49 4E 20 49 47 4E 49 54 49 4F 4E 20 4C 4F 43 4B E2 
[1175866] DEBUG: IBus: RX[16]: 18 0E 68 39 00 82 00 3F 00 07 01 00 01 01 01 FD [SELF]
[1175876] DEBUG: IBus: RX[6]: C8 04 C0 27 00 2B [SELF]
[1175884] DEBUG: IBus: RX[6]: 80 04 30 1B 00 AF 
[1175956] DEBUG: IBus: RX[8]: A4 06 BF 70 02 00 3F 50 
[1178574] DEBUG: IBus: RX[9]: D0 07 BF 5B 00 00 00 00 33 
[1178580] DEBUG: OFF &gt; Unrelated activity!
[1185671] DEBUG: IBus: RX[7]: 44 05 BF 74 04 01 8F 
</pre></div></div></div></div></div></div></div></div><!--EndFragment-->
</body>
</html>
```

closed 2022-01-01 by bmos