tedsalmon/BlueBus #86: Microphone not picking up my voice

nicjansma, 2021-09-08:
This may or may not be related to https://github.com/tedsalmon/BlueBus/issues/78, but I figured I'd open a separate issue in case it's a different cause.

I have a 2002 M3 (mfr date 11/2001).

I installed BlueBus last week, and the Bluetooth functionality is working great.  I can hear music from my phone through the car audio.

When I place a call, I can hear the other end of the line, but they can't hear me.  I've tried calling my wife, as well as an "echo" telephone number that should repeat what I say.  Neither pickup my voice.

I had to purchase and install a microphone for behind the mirror.  It's part number 84318380338 from ECS tuning.  I also purchased the Bluebus SES TCU Adapter ES#4158580 from there as well.

[image]

I'm not sure if it's related, but one note is that the microphone connection didn't perfectly match the car's pre-wired harness.  On the microphone, it has the little "stair pattern" (on the right side):

[image]

On the wire harness, there's an additional "downward edge" that extended out.  It wouldn't fit in the harness because of that extra edge.  In this picture, you can see some deformed plastic on the edge where I took a knife to cut that part out.  It now fits fine.

[image]

[image]

I've confirmed the harness delivers power and ground.

Is there anything I can debug here?

bcagnoni, 2021-09-08:
> [2021-09-08 08 48 43](https://user-images.githubusercontent.com/1004649/132513211-ec691104-1bea-436a-8166-0e11327bd82b.jpg)
> [2021-09-08 08 49 01](https://user-images.githubusercontent.com/1004649/132513270-f3b4f739-ba12-4af2-81d7-ddd93aa6037c.jpg)
> [2021-09-08 08 50 06](https://user-images.githubusercontent.com/1004649/132513456-a8e26014-0ca9-4282-a987-0b22a56cfa50.jpg)
> [2021-09-08 08 50 55](https://user-images.githubusercontent.com/1004649/132513670-5ada0870-5109-47e1-a63c-ffac2f43d504.jpg)

Hello Nic,

when I first installed the BB I had an audio issue. Both myself and the other side couldn't hear anything. I followd the procedure to apply the CVC license and this made the trick.

https://github.com/tedsalmon/BlueBus/wiki/Clear-Voice-Capture

Cheers,
Battista

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

tabor-a, 2021-09-09:
> I'm not sure if it's related, but one note is that the microphone connection didn't perfectly match the car's pre-wired harness. On the microphone, it has the little "stair pattern" (on the right side):

Are you sure that's the microphone wiring and not something else?

tedsalmon, 2021-09-09:
> This may or may not be related to #78, but I figured I'd open a separate issue in case it's a different cause.
>
> I have a 2002 M3 (mfr date 11/2001).
>
> I installed BlueBus last week, and the Bluetooth [...]

Hello @nicjansma!

One important thing to note about these cars is that every connector is color matched to the harness plug and the different key-ways are used to prevent things from being plugged in incorrectly at the factory. In your case as @tabor-a suggested, you have found the homelink connector and that's why it's not working. You need to dig around for the black 3-pin connector.

Thanks,
-Ted

nicjansma, 2021-09-09:
Thank you everyone!

After a lot of searching I finally found the proper black connector microphone cable tucked further into the headliner.  It fits great!  Thanks for the tip.

Unfortunately... my voice is still not being heard by whoever I call.  I can hear the other end's voice no problem through the car speakers.

@bcagnoni mentions with their issue that _both_ sides couldn't hear any audio, so I'm not sure if my issue (where only _my_ mic doesn't work) is related to Clear-Voice-Capture, but I tried to run this: https://github.com/tedsalmon/BlueBus/wiki/Clear-Voice-Capture

It connects find, says _Serial No: NNNN_.  Then when I do run _Apply cVc_, it gets stuck on _Waiting_ and never completes.  I've tried running it multiple times, resetting the BlueBlue in between/etc.  It never seems to complete.  I'm using MacOSX and have tried multiple USB cables.

I'm able to separately terminal into the unit and can see the terminal interface OK.

Are there any debug logs I can enable to figure out if the mic is really picking up my voice?

Or is there a way to tell if CVC already works and is licensed?

tabor-a, 2021-09-09:
Just a guess, but perhaps plugging the HomeLink wiring into the microphone did it in?

tedsalmon, 2021-09-09:
@nicjansma -
> Unfortunately... my voice is still not being heard by whoever I call. I can hear the other end's voice no problem through the car speakers.
As @tabor-a mentioned it's possible that supplying 12v on the ground pin caused issues for the FETs on the microphone.

Can you share pictures of how the BlueBus is wired to the telephone connector in the car?

> Or is there a way to tell if CVC already works and is licensed?
It's unlikely that this is your issue as you're able to hear the other side. Nonetheless, you can check:

```
set log bt on
bt license
set log bt off
```

nicjansma, 2021-09-22:
Picture of the car wiring:

[image]

Here's the license output:

```
~~~~~~~~~~~~~~~~~~~~~~~~~
BlueBus Firmware: 1.1.18
Try HELP or ?
~~~~~~~~~~~~~~~~~~~~~~~~~
# [79555] DEBUG: IBus: RX[7]: 5B 05 80 83 00 08 55
...
set log bt on
OK
bt license
[101850] DEBUG: BT: Send Command 'LICENSE'
OK
# [101870] DEBUG: BT: APTX=
[101881] DEBUG: BT: CVC=3A6B CB90 A9BF FC81 0000
[101885] DEBUG: BT: OK
...
set log bt off
OK
```

So the car's mic cable isn't powered?  Yeah I could see how the homelink might've fried it.

If the license stuff above looks OK, I'll just order another mic.

nicjansma, 2021-10-18:
I picked up another mic from ECS, and unfortunately the other side of the call still can't hear me.

Does the above license stuff look OK?

Is there anything else I can check?  I looked at a couple fuses but didn't see anything obviously blown.

tedsalmon, 2021-11-06:
@nicjansma - Sorry for the lag! The ones in the headliner though! They are connected, though.

-Ted

nicjansma, 2021-11-10:
Thanks @tedsalmon!  As far as I can tell, there is no voltage coming from the headliner mic wires.

I'm trying to determine if that goes through a fuse that might be blown, but haven't found it yet.

tedsalmon, 2021-11-14:
@nicjansma

It should be directly wired to the BlueBus. Any change you can pull the BlueBus from the case and test voltage (with everything connected) at the two microphone pins? You'll see them labeled once you flip the PCB upside down. You should also test from the bottom side of the PCB.

Thanks!
-Ted

nicjansma, 2021-11-28:
@tedsalmon really appreciate you debugging this with me.

I pulled the case off the BlueBus and tested the voltage when everything else was plugged in.  Mic+ and Mic- read ~11V.

However, when I try to read voltage at the mic's wire in the headrest, I never see any voltage across any of the pins.

Is there anything in between the BlueBus and the mic wire in the headrest?  A fuse somewhere?  I looked at the E46 fuse list and didn't see anything about a microphone.

tedsalmon, 2021-11-28:
@nicjansma 

Did you get the 54-pin adapter or the SES adapter?  The only "break" between the headliner and the BlueBus would be the SES connector, where if that's is missing it's jumper, the microphone circuit is broken.

I assume you tested with the adapter connected to the BlueBus, yeah?

-Ted

nicjansma, 2021-11-28:
I got the SES adapter https://www.ecstuning.com/b-bluebus-parts/bluebus-ses-tcu-adapter/ses-tcu-adapter~/

Correct, the SES adapter is connected, though I'll examine that end as well.

tedsalmon, 2021-11-28:
@nicjansma 

Can you send pictures of how everything is connected?

Can I also have pictures of both ends of the SES cable so I can ensure the wires are routed correctly?

-Ted

nicjansma, 2022-05-16:
Sorry for the long delay in responding here, I had the car wrapped up for the winter months.

I debugged this a bit further last weekend.  To start, I ran a continuity test between the microphone's plug/wire in the headliner and the 6-wire coming out of the SES adapter.  I could not find any wires from the plug in the headliner that had continuity with any of the wires in the 6-wire from the SES adapter.

I also looked to see if there was any continuity between ground and any of the wires from the plug in the headliner (by testing against the cigarette plug, as well as against a bolt in the doorway) and could not find any.

So at this point, I'm thinking the plug in the headliner has a break in it somewhere between it and the SES adapter?  I looked at my Bentley manual but couldn't find any obvious wiring diagrams that showed any other junction points after the wire leaves the headliner.

I can get those pictures if you think it'll help, but I'm leaning towards the headliner wires just not being connected to the trunk?

Any ideas?

tedsalmon, 2022-06-29:
Closing issue due to lack of response. Shoot me an email if this is still a problem!

closed 2022-06-29 by tedsalmon