tedsalmon/BlueBus #88: How to recover bootloader after failure during firmware upgrade

vgergiev, 2021-09-15:
Hi Ted and team,
I've made a mistake and ejected usb from bluebus during the firmware upgrade (please don't ask why :( )
Now it's blinking red light and I can't connect to it with  "Error 5: Access is denied".

Is there a way for me to restore bootload and make this magic box work again?

Thank you a lot in advance,
Vlad

tedsalmon, 2021-09-15:
@vgergiev 

Disconnect it from USB, then open the case up. You'll find a jumper near the BC127 labled "RECOVERY". Install the jumper so its in the shorting position then reconnect the BlueBus to USB. The Green light should begin to blink. Remove the "RECOVERY" jumper and install it how it was prior and you'll then be able to flash the firmware.

-Ted

vgergiev, 2021-09-15:
Thank you, I've already figured it out.
Just verified that everything works.
Thank you again for such good design :)

Regards,
Vlad

closed 2021-09-15 by tedsalmon