tedsalmon/BlueBus #90: Feature Request: Angel eyes as DRL

rsporsche, 2021-09-16:
I would like to have the angel eyes in my E39 come on whenever I drive without having to switch the lights manually or run additional wires.

I understand that the 'Intravee II' can do this via IBUS commands, is there any chance this could also be done by the BlueBus?

piersholt, 2021-09-17:
This is in beta.

piersholt, 2021-09-17:
A note regarding terminology. From [M5Board](https://www.m5board.com/threads/fully-integrated-bluetooth-solution.603108/page-42#post-7637320).

> Ted and I were discussing the terminology, which in the case of daytime running lights (DRLs), is a little complicated.
>
> In short, in countries where DRLs are mandatory, the parking lights (angel eyes) do not have sufficient light output to qualify as DRLs. Thus, to call the feature "DRLs" is a potentially risky misrepresentation.
>
> It might seem like a trivial distinction, but when we're talking about vehicles in operation, it's better to err on the side of caution.
>
> Incidentally, this is why there's no "parking lights as DRLs" option in light module coding.

CaptnO, 2021-09-17:
I can confirm that. Even if you replace the parking light bulbs with LED bulbs you won't see them in daytime. Therefore the original DRL uses dimmed highbeams, which are quite easy to be activated via coding. If you intend to use the angel eyes as DRL I suggest to retrofit the power output yourself since the original one is too low.

rsporsche, 2021-09-17:
Ok, I don't mind what it's called, maybe 'Angel Eyes on all the time because they look cool'.

I added LED bulbs and they're bright enough, maybe not modern car DRL bright but then I have no legal requirement where I live for DRLs.

Can I try the Beta? is it stable enough for public consumption?

Should this be closed and linked to another issue?

piersholt, 2021-09-18:
> Can I try the Beta? is it stable enough for public consumption?

You can certainly try the beta, but as with any beta release, there's increased risks of bugs. If stability is a concern, I'd wait for the firmware update.

From [M5Board](https://www.m5board.com/threads/fully-integrated-bluetooth-solution.603108/post-7637191):

> https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_18_beta_3.hex
> 
> I don't know if it works, and I haven't remotely tested this on my car, but I added constant parking lights under Settings > Comfort > Park Lights
> 
> I believe on the E39 this will activate the angel eyes. On other chassis, this will activate the front corner lamps (the lower output setting of the blinker bulbs).
> 
> Also included in this update:
> - MKI / MKII / TV Module detection fixes
> - A2DP volume set to maximum automatically (recommended. This will be configurable in the final release of 1.1.18, but right now it's stuck on).
> - A2DP volume halving when PDC is engaged or the automatic transmission is shifted to reverse. If you don't have a slushbox or PDC, then you're going to continue having to lower your volume as you reverse out of a spot, if that's something you do.
> 
> -Ted

closed 2021-09-18 by piersholt