tedsalmon/BlueBus #92: Mini R53: comfort blinkers do nothing

Sandrowski, 2021-09-22:
I have recently installed a BlueBus in my R53 Mini (Euro model, manufacturing date 09/2005). All the bluetooth functions work just fine, but the comfort blinkers do nothing. Comfort blinkers are set to blink 3 times. The indicators work normally, but a quick tap on the stalk just gives me 1 blink.

The car is equipped with a Siemens made CD53 (Harman/Kardon system)

I've attached a log file (system and iBus). Ignition was in position 2, radio off and phone not paired. After enabling logging the BlueBus was rebooted. After the BlueBus rebooted i did a short single left blink, waited a bit, and then a single short right blink.

[comfort_blink.log](https://github.com/tedsalmon/BlueBus/files/7213156/comfort_blink.log)

piersholt, 2021-09-24:
Thanks for taking the time to capture a log. That's always a great help!

The MINI is somewhat of a mischievous child... For all intents and purposes, it _looks_ like the MINI is equipped with a light module, but after closer inspection it became apparent that there was some shenanigans going on!

We eventually realised that the MINI- despite utilising BMW electronics, doesn't actually have a light module. Given the relative simplicity of the MINI lighting, the lighting electronics were shoehorned into the body module- presumably as it's more cost effective, and in order to maintain compatibility with existing BMW electronics, this Frankenstein body module masquerades as a light module.

You can see this charade (and it's subsequent shortcomings) in the log you provided.

```
[20010] DEBUG: IBus: RX[5]: 3F 03 D0 0B E7 [SELF] # BlueBus requesting light module status
[20019] DEBUG: IBus: RX[5]: 00 03 3F B0 8C # Frankenstein body module replying "NEIN, I KNOW NOTHING!"
```

In short, because there's no light module, we can't use our lighting interface to do the comfort blinker magic.

However, some tentative poking and prodding of the MINI Frankenstein body module suggests we mightn't be out of options, but a good bit of work is required as yet.

I'd recommend disabling comfort blinkers for now.

Sandrowski, 2021-09-25:
Thanks for looking into this issue! 

I suspected that the 'integrated' light module would be the problem. If you need me to test something or provide additional logging or something, please let me know!

piersholt, 2021-09-26:
Thanks for the offer @Sandrowski, that would be a great help come time.

Between the E38, E39, and E46, we're able to test most modules, but there's always a few- like the MINI, that we don't have ready access to. (Although, an R53 Cooper S is on my shortlist! 😄 )

jaripetteri, 2024-02-15:
Any update on this issue?

piersholt, 2024-02-18:
No update. Any energy is being used to sort out the comfort locking features.