tedsalmon/BlueBus #93: No sound in calls, “squeaky sound with RPMs”

hribecek, 2021-09-23:
Hi,

First of all a huge thanks to Ted for making our cars a bit more contemporary with this great device. :)

I have 2 issues with BlueBus in my E39.

1) when I make a phone call neither me nor the other person hear anything, it is just silence.

2) when I listen to music, especially to some classical music which is usually recorded with a bit lower volume - and therefore I need to turn the volume in the car up - there is high pitched "squeaky noise" in the background which is intensifying with RPMs. On idle you can barely hear it, but above 3000RPM it starts to be quite annoying. If I turn the volume down, it reduces as well and it is only present when BlueBus is selected.

My car is EU (delivered new to Germany) 03/2001 530i. Delivered with wide screen MK3 navi, HiFi sound (not DSP) and handsfree for some Motorola phone with a connecter and bay in the armrest.

Before installing BlueBus I already exchanged my navi for MK4 and I had installed the BMW ULF module for Bluetooth handsfree. I also removed the armrest bay for the Motorola and since it also served as a connection unit between cables to the handsfree module in the boot and the mic in front I did connect the pins in those armrest connectors when installing the ULF. I do not remember the exact pins as I did that maybe 7 years ago, but the ULF calling worked fine (apart being a bit low volume from the speakers, which I believe is normal for cars without DSP).

Any suggestions would be very appreciated.

tedsalmon, 2021-09-24:
Hello @hribecek!

Your telephony issue sounds like you're missing the clear voice capture license. Can you follow this guide? https://github.com/tedsalmon/BlueBus/wiki/Clear-Voice-Capture

As per your RPM depends high frequency A/C voltage whine, that sounds like you'll need a ground loop isolator to overcome it. Can you shoot me an email about this? support@bluebus.dev

Thanks!
-Ted

hribecek, 2021-09-24:
Hello Ted.

thank you for your reply. I downloaded the CVC Tool and I will try it in the evening.

OK, I will write you an email in a moment.

Tomas

hribecek, 2021-09-24:
Hello Ted,

Just to let you know, the CVC tool helped, handsfree is now working! Thanks

Tomas

closed 2021-11-14 by tedsalmon