tedsalmon/BlueBus #94: STL file mounting bosses / PCB mounting method

ShonkyCH, 2021-09-24:
In the current design the mounting boss in the top and bottom pieces of the enclosure are more than 1.6mm apart which you'd expect to hold a regular PCB. What is the expected mounting method? I printed the two parts and the base supports seem the right height but the top supports are about 8.4mm off the top of the board i.e. looks like exactly 10mm between the two with the board missing.

Am I missing something with the design? I could probabably stretch the bosses in the top section but am I missing something? Or do you just screw the board into the base and the top bosses don't do anything? The PCB mounting holes are a bit small for a reasonable screw size and the boss holes are too big as is.

I use mine all the time just for the audio playback and it really does work well. Thanks.

tedsalmon, 2021-09-24:
> In the current design the mounting boss in the top and bottom pieces of the enclosure are more than 1.6mm apart which you'd expect to hold a regular PCB. What is the expected mounting method? I [...]

I think the only part that you're missing is that there are four heat-sert threaded inserts that are pressed into the bottom case boss holes. There are four M2 screws that sandwich the case and PCB.

https://www.mcmaster.com/94459A120/
https://www.mcmaster.com/92010A111/

There's an updated design that's way better -- https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/BlueBus+(ThreadForming)+v5.f3z (I don't have STLs available yet).

The new case uses this screw: https://www.mcmaster.com/99397A267/

>The PCB mounting holes are a bit small for a reasonable screw size and the boss holes are too big as is.
I wanted small screws at first, but now I realize that was inflicting more pain on myself than was really necessary. I'll be laying the BlueBus PCB back out in the coming months to replace the BC127 and I will likely expand the mounting hole size at that point to 4mm diameter.

ShonkyCH, 2021-09-24:
Thanks. Yeah I figured out you might be using the heat serts after my post because the base holes are 3mm. With a 25mm bolt, that holds the top to base plastics together no problem, but how is the PCB held against the base? The bosses from the top plastic are too far above the PCB. I could print or use some spacers for mine or extend them an print another.

This gap. Ignore the dimension (it's 10x) as Fusion 360 gets the untis wrong when importing the STL and it's not a straight face to face measurement
[image]

Your new design has the bosses from the top extending to the PCB which sandwiches the PCB between the two pieces of plastic which is what I was expecting. Much cleaner with the screws from the bottom and no nutserts required.

tedsalmon, 2021-10-03:
@ShonkyCH,

>but how is the PCB held against the base? 

It's not ;). It made assembly... fun. There's also the issue of the SMB connector not allowing the board to sit flush, which was fixed in the redesign.

See, someone (a user) who makes enclosures for a living created v2.0 -- v1.0 was me sketching a design and having a buddy mock it up in F360, then having another guy tweak it a bit more to make it a viable print.

-Ted