tedsalmon/BlueBus #96: Request: Comfort lock only unlock the drivers door

calmelb, 2021-10-01:
Unsure if it's possible but it would be nice if the comfort locks could just have an unlock feature (for cars that already have it) and only unlock the drivers door (since it's better for safety). 

I have a BMW E46 sedan

tedsalmon, 2021-10-03:
@calmelb,

The issue here is that we start creating different features for different chassis -- I may be wrong but the last I remember there wasn't a good command to just unlock the high side doors on the E39, but there is on the E46.

For the E39, I use the lock/unlock button diagnostic request to lock/unlock the doors, for example.

I'll go digging through the EDIABAS command dump sometime this week and see if there's anything that can be done for all ZKEs.

-Ted

calmelb, 2021-10-04:
Thanks @tedsalmon! Appreciate it. Understand about the differing features, still would be a neat feature though since I prefer only having my door open rather than all. 

Also what about at least having an only unlock option since my car already automatically locks? Even if it unlocks all the doors